sebanc/brunch #1198: Brunch Framework r94 (Pre-Release)

admin-elvistrujanovic, 2021-08-12:
Can you add r94 pre-release version for 94.0 Chrome OS dev channel support? I'm on that version but I can't login to my chrome os anymore after that upgrade. I have r91 latest release but it's not working for that version.

WesBosch, 2021-08-13:
Typically brunch is forward compatible with releases of ChromeOS with few exceptions and releases coincide with stable versions. 

It's generally suggested to stay on the stable channel, as the other channels have too many bugs and change too frequently to make releases for. Brunch 94 probably won't happen until ChromeOS hits the stable channel.

admin-elvistrujanovic, 2021-08-13:
> Typically brunch is forward compatible with releases of ChromeOS with few exceptions and releases coincide with stable versions.
>
> It's generally suggested to stay on the stable channel, as the other [...]

Thanks for answer,

I found a solution for my problem - simply, I started the Powerwash process and everything it's fine

but

Does that also explain the fact that I can't run the crostini and Android apps on the dev channel?

WesBosch, 2021-08-13:
that's likely just a bug in the dev channel. Crostini has been somewhat broken on the last few releases on stable. Android apps are equally buggy as arcvm is being tested. If you're using the dev channel, expect bugs. that's what it's for.

admin-elvistrujanovic, 2021-08-14:
> that's likely just a bug in the dev channel. Crostini has been somewhat broken on the last few releases on stable. Android apps are equally buggy as arcvm is being tested. If you're using the dev channel, expect bugs. that's what it's for. 

👍

closed 2021-08-14 by admin-elvistrujanovic