sebanc/brunch #1202: risks of running on non-certified hardware?

RJVB, 2021-08-19:
Hi,

I found this interesting project via https://beebom.com/how-install-chrome-os-on-pc/ . The author suggests to use a secondary gmail account to set up the system after installing when not running on hardware that has been certified by Google.

Is there any substance to the implied risk that your google account might be suspended for being used on an "illegal" ChromeOS install?

khanhtranngoccva, 2021-08-30:
probably not. Many developers and power users use custom ROMs all the time. You don't really need to worry until you put devices for sale.

RJVB, 2021-08-30:
Thanks, that's what I guessed (but one never knows if one is paranoid enough ;) )

closed 2021-08-30 by RJVB