sebanc/brunch #1207: editing boot options (or booting into debug mode) means doing the setup again?!

RJVB, 2021-08-21:
I'm getting the feeling that each time you change the options in the GRUB configuration you're also going to have to go through the entire initial setup again? Which probably also means the same will happen if your debug boot configuration is different and you have a reason to boot into that mode.

That's not something anyone must like to discover or go through, so it'd be nice to address this in the instructions!

sebanc, 2021-08-22:
Changing options or debug mode are not supposed to bring you to the initial setup.

Kernel change would not either but might require you to do a powerwash (which is documented in the readme).

I suspect you are trying to use a brunch usb on different machines which will actually bring you to the initial setup due to chromeos security rules.

RJVB, 2021-08-22:
On Sunday August 22 2021 01:06:55 sebanc wrote:

>I suspect you are trying to use a brunch usb on different machines which will actually bring you to the initial setup due to chromeos security rules.

I did that too, once, and concluded it was a bad idea (too). But I definitely have been doing the initial setup dance so often that I'm about to give up on the whole OS because it's been taking me way too much time. The machine I wanted to test it (a Chuwi Hi10 Air) on is excruciatingly slow (hence the idea to test-drive an OS tailored for low-end hw) and apparently the thumb drive I picked for this experience (https://www.amazon.fr/gp/product/B0749XGV86) isn't much faster than a regular old thumb drive despite the claims. I also have the impression that not all OS settings are backed up so after each powerwash I have been going over them.

BTW, is it normal that the rootfs gets rebuilt each time you change an option? It seems to me, but maybe that triggers the initial reset?

FWIW, I downloaded chromeos_13904.55.0_rammus_recovery_stable-channel_mp-v2 (strange download site for those images btw; it mentions stable/canary/dev versions that don't seem to correspond to the downloadable images).

sebanc, 2021-08-22:
> I did that too, once, and concluded it was a bad idea (too). But I definitely have been doing the initial setup dance so often that I'm about to give up on the whole OS because it's been taking me [...]

ChromeOS is not really made to be run on a usb flashdrive, even fast.

> BTW, is it normal that the rootfs gets rebuilt each time you change an option? It seems to me, but maybe that triggers the initial reset?

Yes it is normal, but usually it takes a few attempts to find the correct options and you never have to change them again.

> FWIW, I downloaded chromeos_13904.55.0_rammus_recovery_stable-channel_mp-v2 (strange download site for those images btw; it mentions stable/canary/dev versions that don't seem to correspond to the downloadable images).

"CrOS Updates Serving" is a website managed by Google and https://cros.tech/ is just a differently formatted mirror managed by the ChromeOS community as "CrOS Updates Serving" is often down. The stable / beta / dev versions of the board are there for information but are not related in any way to the recovery image.

RJVB, 2021-08-22:
On Sunday August 22 2021 02:31:10 sebanc wrote:

>ChromeOS is not really made to be run on a usb flashdrive, even fast.

I should have gotten one of those flashdrives that are more like SSD drives in a thumb drive format, but fell for the smaller, cheaper format of the Verbatim that also has a USB-C connector.

>> BTW, is it normal that the rootfs gets rebuilt each time you change an option? It seems to me, but maybe that triggers the initial reset?
>
>Yes it is normal

I thought so, but could it be that there's something in recent OS versions (or your own scripts) that trigger a powerwash each time the rootfs is rebuilt?

sebanc, 2021-08-22:
> I should have gotten one of those flashdrives that are more like SSD drives in a thumb drive format, but fell for the smaller, cheaper format of the Verbatim that also has a USB-C connector.

You should probably install on hdd if you have enough storage, note sure how it would run on a ssd through USB (probably a big limitating factor depending on the device capability)

> I thought so, but could it be that there's something in recent OS versions (or your own scripts) that trigger a powerwash each time the rootfs is rebuilt?

As far as I can tell you are the only one having this issue, not sure what's wrong. Maybe some virtual tpm files are corrupt, you could try a clean install.

RJVB, 2021-08-22:
>You should probably install on hdd if you have enough storage, note sure how it would run on a ssd through USB (probably a big limitating factor depending on the device capability)

The internal is a non-replaceable 64Gb eMMC and for now I'm keeping its MSWin10 install - I wouldn't want to try how "fast" that OS will be from a slower device and its the only one that has drivers for the touchscreen and rotation sensor :(

An SSD takes very little performance hit over USB3, at least not one that's meaningful on lower-end consumer hardware.

RJVB, 2021-08-22:
Well, I dd'ed the content of the Verbatim drive to a Samsung Fit, and
- I did get a rebuild of the rootfs, presumably because of the additional partition filling the remaining 30-some Gb, but *no* powerwash.
- behaviour is much better now, and the device even feels faster than it is under Linux (Devuan Beowulf). We'll see if that feeling remains once I start doing my once-a-day Facebook scans ;)
- the Play Store app now works 
- Setting up the Linux environment still fails during the VM download, without giving a reason

sebanc, 2021-08-22:
There is a bug with crostini in chromeos r91, you can try to disable this flag and install again:
chrome://flags#crostini-use-dlc

However, there is also a minimum speed needed to get crostini working and it won't pass the install on most USB flashdrives because of this.