sebanc/brunch #1210: Brunch stuck on patches being applied

ayoub99-max, 2021-08-22:
I dual booted chrome os every thing was fine until I booted from grub it showed me a lot of read only errors and then it ended up by self repair

ayoub99-max, 2021-08-22:
[image]

ayoub99-max, 2021-08-22:
[image]

ayoub99-max, 2021-08-22:
[image]

ameyjoshi3068, 2021-08-23:
I would suggest you to first make sure your device is compatible
If you have installed Chrome OS on windows partition (NTFS) then just make sure you have disabled fast startup and completely shutted down windows.