sebanc/brunch #1211: Alternative dual boot method proposal: Nested GPT

hacker1024, 2021-08-23:
It's possible to create a GPT partition table inside a regular partition. For example:

```
/dev/sda: GPT table for disk
  /dev/sda1: ESP
  /dev/sda2: Windows
  /dev/sda3: GPT table for Chrome OS
    /dev/sda3p1
    ...
    /dev/sda3p12
```

I've done some minimal testing, and it looks like both GRUB and Linux deal with this properly.
There are two problems with the current disk image approach:
1. It's not very resilient; if there's a loss of power or kernel panic, there are two filesystems that could get corrupted (the partition filesystem, and filesystems in the disk image)
2. It's not as performant as all I/O needs to go through an additional filesystem driver (NTFS-3G is particularly slow in some cases)

I've taken a look at the init script, and I don't think it will be too hard to support this dual boot method. I can even attempt it myself. I'm not sure how the Chrome OS updater will handle this, though - does anyone know if it will be okay?

hacker1024, 2021-10-06:
@sebanc I'd love to hear your thoughts on this

hacker1024, 2021-10-07:
I'm not too sure what you mean by modifying existing partition tables, unless you're referring to the creation of a single partition on a disk to hold all the sub-partitions. Chrome OS's own partition table remains untouched.

I would not have the install script modify any tables at all, though - I was thinking more along the lines of it detecting when `dst` is a partition instead of a disk, and then using this method, much like how it works with disk images.

sebanc, 2021-10-08:
The above change is included in the latest brunch-unstable build:
https://github.com/sebanc/brunch-unstable/releases/tag/r93-unstable-20211008

hacker1024, 2021-10-09:
Looks good to me, I'll test it in the next 7 days.
Personally, I'd nest the `img_path` check in a single `img_part` check for performance and readability, but that hardly matters.

hacker1024, 2021-10-09:
Oh, slight problem - `img_part` will not be set if `img_path` is missing.

hacker1024, 2021-10-11:
The loop setup logic is identical between nested partitions and images as well.
To remove code duplication, could something like this be done?

```bash
if [ -e "$img_part" ]; then
	[ ! -z "$img_path" ]; then
		mkdir /mainroot
		fstype=$(blkid -s TYPE -o value "$img_part")
		if [ "$fstype" == "ntfs" ]; then
			ntfs-3g "$img_part" /mainroot
		else
			mount -n "$img_part" /mainroot
		fi
		full_img_path=/mainroot/"$img_path"
		if [ ! -f "$full_img_path"]; then
			echo "brunch: ChromeOS loopfile $img_path not found on device $img_part..." > /dev/kmsg
			full_img_path=
		fi
	else
		echo "brunch: Trying to boot in nested GPT mode on device $img_part..." > /dev/kmsg
		full_img_path="$img_part"
	fi
fi
if [ ! -z "$full_img_path" ] then
	mknod -m660 /dev/loop15 b 7 480
	losetup -P /dev/loop15 "$full_img_path"
	bootdevice=/dev/loop15
	partpath=/dev/loop15p
else
...
```

xAffan, 2021-12-08:
Dude come on give some updates. This is interesting and a lot better than using IMG's

sebanc, 2021-12-08:
As mentioned in my previous posts, I don't share this opinion but that is just my point of view.

I made brunch compatible with this technique and this is as much as I will do.

Now that brunch is compatible with this technique, someone just has to make a custom install script which makes use of it.

hacker1024, 2021-12-16:
What happened in R96? I'm most concerned about this bit:
> Moreover, updating a dual boot install to ChromeOS r96 will prevent you from accessing your user data. Therefore, make sure you have a backup before updating.
What changed to cause this? Is this dual-boot method also affected?

hacker1024, 2021-12-16:
I've just tried to set this up on a new Ryzen laptop (Dell Inspiron 7415) for a family member.
Interestingly, its UEFI firmware doesn't seem to be recognising the nested GPT table. A single MBR partition shows up under the GPT partition instead of the usual 12 GPT partitions.
Does anyone know what part of the UEFI firmware handles partition mapping? If it's just a driver, I'd like to try loading the one from my laptop's UEFI firmware on this new one.