sebanc/brunch #1216: GPU Acceleration

rzrbg, 2021-08-30:
I have Prestigio 133s-N3350,3GB RAM,HD500.It work flawlessly with Coral and Brunch 90 !This netbook its just made for Chrome OS.. thanks to you!But there is some strange problem with games.Roblox is so so laggy even in the menus(totally unplayeble),Among Us is laggy aswell,so I think its something related to GPU or Android container.Hyper Treading is enabled what else I can do?I`ve tried rammus and coral.Coral chromebooks are 95% same specs like this one.Any suggestions are welcomed.

Andytheclown, 2021-09-13:
I’m only speaking from experience and trying to get Roblox and AmongUs working on a PC. They’re quite graphically intensive. If we look at Roblox, it really does require some quite good GPU acceleration to run, even in lower resolutions… under Windows its slow with onboard intel graphics…. So it’s not going to be much different with ChromeOS

rzrbg, 2021-09-14:
In Windows 10 it works like a charm... both games with no lags or whatever. I've tried FydeOS and there is no such problem.