sebanc/brunch #1220: Silead touchscreen calibration parameter "min_x" not working as expected

jlempen, 2021-09-03:
Hi sebanc, hi all,
Thanks for your fantastic work on the brunch project!

My Chuwi SurBook Mini 2-in-1 tablet has a Silead touchscreen which works without any need for calibration in Arch Linux with kernel 5.4 or 5.10 once I copy the firmware file from [onitake's git repo](https://github.com/onitake/gsl-firmware/blob/master/firmware/linux/silead/gsl1680-chuwi-surbook-mini.fw) to `/lib/firmware/silead/`.

On brunch r91 with the `rammus` recovery image and kernel 5.4 or 5.10, the Silead touchscreen starts working as well once I copy the same firmware file to `/lib/firmware/silead/`, but unexpectedly exhibits an offset of about 100 pixels on the x-axis. This seems rather strange, as brunch uses the same `silead_ts` linux kernel module and definitions for my tablet in the `touchscreen_dmi.c` source file:

```
static const struct property_entry chuwi_surbook_mini_props[] = {
	PROPERTY_ENTRY_U32("touchscreen-min-x", 88),
	PROPERTY_ENTRY_U32("touchscreen-min-y", 13),
	PROPERTY_ENTRY_U32("touchscreen-size-x", 2040),
	PROPERTY_ENTRY_U32("touchscreen-size-y", 1524),
	PROPERTY_ENTRY_STRING("firmware-name", "gsl1680-chuwi-surbook-mini.fw"),
	PROPERTY_ENTRY_U32("silead,max-fingers", 10),
	PROPERTY_ENTRY_BOOL("touchscreen-inverted-y"),
	{ }
};

static const struct ts_dmi_data chuwi_surbook_mini_data = {
	.acpi_name      = "MSSL1680:00",
	.properties     = chuwi_surbook_mini_props,
};
```
The offset I see on my device looks very much like the "touchscreen-min-x" value of 88 in the above file. Maybe I also have an offset of 13 on the y-axis, but this value is too small to be determined precisely anyway.

Now, I'm trying to compensate for this offset by using the brunch kernel parameters for calibrating Silead touchscreens described in the thread [https://github.com/sebanc/brunch/issues/1108#issuecomment-860004205](https://github.com/sebanc/brunch/issues/1108#issuecomment-860004205), with the following parameters on the kernel command line in order to override the above value of 88: `of_touchscreen.custom=1 of_touchscreen.min_x=1`

This produces an unexpected result: on the left side of the touchscreen, I still have the same offset of about 88, but this offset becomes smaller and smaller the more I move towards the right side of the touchscreen, where the offset is zero and the touchscreen perfectly usable.

A value of 0 for `of_touchscreen.min_x` seems to have no effect and a negative value of -88 gives me the same offset of 88 on the left side of the touchscreen, zero offset in the center and a reverse offset of -88 on the right side of the touchscreen.

So, the parameter seems to work as expected on the right edge, but has no effect at all on the left edge of the touchscreen, which makes it utterly useless for calibrating a Silead touchscreen, or at least the touchscreen of my device.

@sebanc, do you think this could be due to a bug in the code for overriding the `silead_ts` module parameters in your source file `of_touchscreen.c`?

Thanks for having a look at it!

jlempen, 2021-09-06:
Update:

I downgraded my install to Brunch r87 and rammus r87, which was the last release before Sebanc added the Silead touchscreen calibration parameters.

My Silead touchscreen has the same offset issue in r87 as well. This confirms that the issue is not due to a bug in the override code in the source file `of_touchscreen.c`.

I still cannot figure out why my touchscreen has this annoying offset and why I'm unable to correct the offset with the override kernel parameters `of_touchscreen.custom=1 of_touchscreen.min_x=1`.

sebanc, 2021-09-06:
Sorry I cannot really help with this as I do not have such device, my thoughts are:
- The kernel is probably not related to this issue as you noticed, I changed nothing else than the small parameters override which should not cause that.
- 0 value will not work when you override parameters but 1 is probably working as expected, it is just that the issue is somewhere else.
- My guess would be that silead touchscreen parameters are really not common and the chromeos touchscreen driver does not know how to handle them correctly. I would try to make a completely different calibration for chromeos starting with modifying the touchscreen size.

jlempen, 2021-09-08:
Thanks @sebanc!

I'll play around with the calibration settings. 

Another approach would be to calibrate the touchscreen using `xinput` and a `libinput` "coordinate transformation matrix" in the Crostini Linux container and then copy the matrix over to the Brunch install.

Weddin, 2021-10-14:
Also you can use glsx680_acpi_ts. It support to converted firmware. For more information https://github.com/onitake/gsl-firmware

LucaCraft89, 2025-05-18:
Same device and same issue here with fyde os, i have been banging my head i tried usign xinput calibrator in the linux vm and then copying it to `/etc/gesture/99-touchscreen.conf` as follows but nothing
```
/etc/gesture/99-touchscreen.conf :
Section "InputClass"
        Identifier      "Silead TS calibration"
        MatchDevicePath "/dev/input/event*"
        MatchProduct "silead_ts"
        Option  "MinX"  "2349"
        Option  "MaxX"  "68886"
        Option  "MinY"  "282"
        Option  "MaxY"  "65407"
        Option  "SwapXY"        "0" # unless it was already set to 1
        Option  "InvertX"       "0"  # unless it was already set
        Option  "InvertY"       "0"  # unless it was already set
EndSection
```
also tried in the kernel with no result
```
input-core.custom=1 input-core.invert_x=1
```
```
of_touchscreen.custom=1 of_touchscreen.min_x=1
```