sebanc/brunch #1228: My touch screen is inverted-X

Mahavirocana, 2021-09-15:
hello!
Today I finally install Chrome OS successfully r91, but I realized touch screen is inverted -X
For example, if I touch buttom left, It recognizes buttom right. 
I'm newbie for linux/chrome OS, so I don't know anything about kernerlling. 

My tablet is XD10 duo, which is OEM version of "Cube Iwork10", and It seems that Chuwi Hi10 Pro also has simillar spec. (Cherry trail z8300, 4gb ram, 64gb mmc etc.)

[TP_XD10DUO.zip](https://github.com/sebanc/brunch/files/7167927/TP_XD10DUO.zip)

I uploaded touch Calibration driver for windows 10 (Sadly, there are no linux version), so I hope it might help to resolve this problem.

Mahavirocana, 2021-09-15:
@czyun44  [Yes, It works after placing "of_touchscreen.custom=1 of_touchscreen.invert_y=1" after "cros_debug" and before "options=". ]

I followed this and it runs perfectly right now. thank you.

closed 2021-09-15 by Mahavirocana

xenoxc, 2022-02-27:
I am trying Cube iWork10 Ultimate for Chrome OS. I've cleared the Touchscreen problem. but I have the Audio Problem now. It's no sound from speaker
Do you have any audio problem. Can you tell me the kernel you used ?