sebanc/brunch #1234: CrOS Updates 93

Xerrious, 2021-09-25:
Hey guys!

Today I noticed that you cannot download older versions from https://cros-updates-serving.appspot.com/
And those that you can download are version 93, like rammus, grunt and zork... they are all at 93 stable and 96 dev or something like that. The Brunch latest release is 91, so my question is, will it still work? brunch 91 with rammus 93, for example? thanks, cheers!

AbrarYNWA, 2021-09-27:
does rammus 93 work with brunc 91?

sortedcord, 2021-09-27:
Yes

ZZZXXX0110, 2021-10-01:
But this is still not ideal, because Google introduced changes on 93 that broke accelerometer support for Brunch on Microsoft Surface devices and probably many others too, sebanc is working on updating his custom drivers at the moment but it will take him sometime.

I currently have 91 installed on my Surface Pro 6 but it would be problematic for any new comer who wishes to try out Brunch on their Surface devices.