sebanc/brunch #1238: Asus Vivobook E200HA Linux apps not downloading

Eclipse-5214, 2021-09-28:
whenever I try to install Linux app support on my computer it says error could not download pleas help.

peter-marshall5, 2021-09-28:
Is Virtualization Technology enabled in the BIOS?

Eclipse-5214, 2021-10-05:
It's not supported

peter-marshall5, 2021-10-05:
Linux support on Chrome OS requires virtualization technology. It might be hidden in the BIOS menu somewhere or you might need to use a keyboard combination to make it appear. It's possible that virtualization technology is already enabled by default but you might need to add "disablevmx=off lsm.module_locking=0" to the kernel options to use it.

Eclipse-5214, 2021-10-05:
How would I do that?

peter-marshall5, 2021-10-05:
I'm not sure about enabling virtualisation technology in the BIOS on your model of laptop. It's possible that the CPU doesn't support it.

You can ignore the part about the kernel options because they should be already added by default.