sebanc/brunch #1239: [11th gen] Graphical issues with XPS 13 9310 (i7-1185G7)

DavidIlie, 2021-09-30:
Hello everyone, 

I recently bought a new XPS 13 9310 featuring the 11th gen CPU and I managed to get chromeOS installed just fine but there are some problems with the video driver.

- Every time I switch tabs, open a new window, etc. There is a chance the the screen will freeze for a moment or will not update the screen with the new page, leaving segments of the old page on the screen. (I will try and get a video of this later today)
- As with most recent laptops, the microphone doesn't work properly
- I saw using `lsusb` that there is a `Goodix Fingerprint Sensor` that is present in the laptop, but doesn't show up in settings.

I am mainly interested about fixing the graphics issues but that might just be a problem with the kernel that is running in brunch. (I am using 5.10 with a volteer image).

Any help would be appreciated!

0x15f2d, 2021-10-04:
You're in luck man. 11th gen i3-1115G4 here. iGPU rendered fine but when it comes to displaying actual Chrome OS's content. It will blank out the screen with the Mouse cursor left. Sometimes it will statically display pictures but in a very short period of time.

Try switching Developer console back and forth really quick (Ctrl + Alt + F1/F2). It will force the GPU render something.

DavidIlie, 2021-10-04:
> You're in luck man. 11th gen i3-1115G4 here. iGPU rendered fine but when it comes to displaying actual Chrome OS's content. It will blank out the screen with the Mouse cursor left. Sometimes it will statically display pictures but in a very short period of time.
> 
> Try switching Developer console back and forth really quick (Ctrl + Alt + F1/F2). It will force the GPU to render something.

The way I resolve it is alt-tabbing again just to force it to refresh. But I am looking for a way to resolve these issues, as 60% of the time graphics work just fine. However, I noticed that in some pixels on the screen (on pages such as Google Classroom) the colors are just a solid color such as green or black. What I found weird is that Chromebooks with my CPU already exist and they use the volteer image, and graphics works just fine.

Might this be a Dell Firmware issue? Dell is known for having custom firmware for the hardware.

u9wy, 2021-10-05:
@DavidIlie I have the same laptop as you. I've experienced this freezing issue but very very rarely and it resolves its self after a few seconds.

To fix you microphone issue there is kernal option you can use you just need to edit the grub configuration i assume you also have an issue with you camera not working add the following to options in you grub config: internal_mic_fix,no_camera_config

I've added my grub config below.

```
menuentry "ChromeOS" --class "brunch" {
        linux ($disk,7)/kernel-5.10 boot=local noresume noswap loglevel=7 cros_secure cros_debug options=enable_updates,internal_mic_fix,no_camera_config,pwa \
                console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
        initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}
```

I'm wondering if you also have issues with battery life? I seem to be getting at best 5 hours but most of the time 3 hours

DavidIlie, 2021-10-05:
@u9wy Thanks for the grub config, I'll check it out. Perhaps a problem with my install could be that it's an .img? That's the way I have chromeos installed.

Regarding battery life, I get around 5-6 hours when doing basic chrome duties but if I have Linux Mode (beta) on, I get around .

u9wy, 2021-10-05:
@DavidIlie 

> @u9wy Thanks for the grub config, I'll check it out. Perhaps a problem with my install could be that it's an .img? That's the way I have chromeos installed.
> 
> Regarding battery life, I get around 5-6 hours when doing basic chrome duties but if I have Linux Mode (beta) on, I get around .

Hmm perhaps the .img install could be a contributing factor. I wiped my whole drive and installed chome os the normal way. I also have linux enabled and android enabled.

DavidIlie, 2021-10-05:
@u9wy 

> @DavidIlie
>
>
> Hmm perhaps the .img install could be a contributing factor. I wiped my whole drive and installed chome os the normal way. I also have linux enabled and android enabled.

But how could the .img be a factor to this? Before I got this laptop I used to use a HP Spectre X360 with a 10th gen CPU with the exact same method that I had with this laptop, and no problems were present.

u9wy, 2021-10-05:
> @u9wy
>
>
> But how could the .img be a factor to this? Before I got this laptop I used to use a HP Spectre X360 with a 10th gen CPU with the exact same method that I had with this laptop, and no [...]

I suppose you're right it was just a suggestion. I think its more likely that the issue could be to do with the kernal or the volteer image.

DavidIlie, 2021-10-05:
> I suppose you're right it was just a suggestion. I think its more likely that the issue could be to do with the kernal or the volteer image.

The odd thing is that there are real chromebooks using the exact same CPU and are running the volteer build just fine. Might be a kernel issue.

DavidIlie, 2022-01-08:
> Why don't you consider switching to the hatch image(best for Gen11, but sometimes needed kernel 5.10) instead of the volteer(Gen10 or 11)? Set the kernel to 5.10, and enable_updates, then upgrade your OS up to date (Now is Version 97 of official build). Don't try to use Beta or Dev OS because it's unstable under bunch 94. Don't try to use unstable Brunch rs96, it can't work well right now!

Never heard about the hatch image, I will try it out now and see if performance is improved. However, sebanc suggested using the boot argument `oled_display` which resolved most of the screen tearing but I will try this image and see if it is gone completely. Thanks!

xuhongkai194, 2022-01-14:
Oh, sorry to hear about it. My HP laptop ran with the Gen10 1065g7 and can run well both with the hatch and volteer mage on the SD card. I found it's more stable than Dualboot installations on the hard disk. I build the chromeos.img the brunch R97(unstable) and the latest volteer or hatch recovery image. The kernel was set to5.15. 
And used the Framework Options parameter as follows:
1."enable_updates"   2."pwa" : it can enable the brunch PWA for update conveniently
3."mount_internal_drives" 4.rtbth": I used it to solve my Bluetooth faults
5."no_camera_config": I used it to solve my camera faults 6. "suspend_s3".
7."enforce_hyperthreading=1" 8."i915.enable_fbc=0 i915.enable_psr=0"
9."psmouse.elantech_smbus=1" 10."psmouse.synaptics_intertouch=1"
Then After starting, update the os to date, and open the website: chrome://flags#scheduler-configuration, To turn on Hyper-Threading
Finally, use apt install to set up the app that can't be supported by google play. Now, the Chrome OS is almost perfect run in the hp laptop except for the "live transcribe" working. Just for your ref., good luck!

xuhongkai194, 2022-01-14:
Oh, sorry to hear about it. My HP laptop ran with the Gen10 1065g7 and can run well both with the hatch and Volteer image on the SD card. I found it's more stable than Dualboot installations run on the hard disk. I build the chromeos.img with the brunch R97(unstable) and the latest Volteer or Hatch recovery image. The kernel was set to 5.15 by me.
And I used the Framework Options parameter as follows:
1."enable_updates"       2."pwa" : it can enable the brunch PWA for update conveniently
3."mount_internal_drives"     4.rtbth": I used it to solve my Bluetooth faults
5."no_camera_config": I used it to solve my camera faults        6. "suspend_s3".
7."enforce_hyperthreading=1"       8."i915.enable_fbc=0 i915.enable_psr=0"
9."psmouse.elantech_smbus=1"          10."psmouse.synaptics_intertouch=1"
Then After the setup and first starting, update the OS to date, open the website: chrome://flags#scheduler-configuration to turn on Hyper-Threading, and use apt install to set up the app that can't be supported by google play. 
Now, the Chrome OS is almost perfect run in the hp laptop except for the "live transcribe" working. 
Just for your ref., good luck!

DavidIlie, 2022-01-14:
That's a similar setup to what I have now, I am running it in a dual boot format as I still need access to Windows. It works without any problems now.

xuhongkai194, 2022-01-15:
Oh, sounds good! I met a problem with the  "Live Transcribe" App now.
How is the "Live Transcribe" APP running on your laptop?  Why doesn't work on my HP laptop? 
Do you install any other specific App or make any adjustments to run the App normally?
Sebanc said that it might be solved by disabling Dev mode, but I haven't tried it.  Because I will purge all of the user data on my disk. Do you have any better method for it?

DavidIlie, 2022-01-15:
Never used "Live Transcribe" before, but from what I can understand from its name, is that it needs to use the microphone? My laptop's microphone does not work under ChromeOS under any circumstance, even with the microphone option argument or disabling dev mode or other things I've tried throughout the past 2 years of using brunch.