sebanc/brunch #1241: Changing Boot Animation

dtadptvl, 2021-10-01:
Can I change the boot animation from showing "Loading Brunch Framework" to something more attractive? I'm a noob please send help

GerardoHD, 2021-10-02:
[Here are instructions on how to view the available bootsplash in current release ](https://github.com/sebanc/brunch#bootsplashes)

You'll just have to change the "brunch_bootsplash=default" option when booting grub, later;
Press CTRL+ALT+T and type "shell" at the invite, and execute
sudo edit-brunch-config
Then change "brunch_bootsplash=default"  to whatever bootsplash you liked.
Be careful or you could break your system

Hope it helps you

dtadptvl, 2021-10-03:
[image]
I followed your instruction and encounterd this error

dtadptvl, 2021-10-03:
Never mind I got it resolved. Thank you

closed 2021-10-03 by dtadptvl