sebanc/brunch #1252: brunch/grub config updater not working

poly3dfree, 2021-10-06:
I installed the latest brunch r93 with CrOS r93 rammus image. The android container was not starting so I seen this thing in wiki saying android_init_fix. But when I run sudo edit-brunch-config and select stuff it doesn't apply and give me this error.

Summary of the configuration:
options: android_init_fix
kernel: 5.4
kernel commandline parameters: enforce_hyperthreading=1 
verbose mode: 0
brunch bootsplash: default_dark

cp: cannot stat '/mnt/stateful_partition/unencrypted/brunch_config/efi/boot/template.cfg': No such file or directory
Brunch configuration saved.
Press any key to reboot your computer...

after rebooting problem persists so I think it is not applying the configuration could you please confirm.
Same thing when selecting ChromeOS(setiings) from grub.

I'm using this in MBR pc with mbr_support patch.

sebanc, 2021-10-06:
Indeed, I forgot to add one file.

I have just updated the patch, could you try the new one ? (if you know how to do it, you can use "dd" to copy mbr.img over partition 12)

poly3dfree, 2021-10-06:
Hello sebanc,
Now although its not showing error but apps are still not opening I get infinite loop of app's loading symbol it just keeps revolving and doesnt open.
Could you please tell how can I confirm if changes are done.
Before we could just edit boot file with options, could you tell it's location so I can see? I was unable to find that file. It would be great.

sebanc, 2021-10-06:
Restart chromeos, log-in then wait 2 minutes and post a dmesg (run "dmesg > ~/Downloads/dmesg.txt" and post dmesg.txt from your downloads folder.

A dmesg is the only way to see if the options... are correctly taken into account and from there I can debug it.

sebanc, 2021-10-06:
android apps do not work ? the log looks pretty good.

Which recovery image are you using ?

sebanc, 2021-10-06:
*and everything seems ok regarding the options setup

poly3dfree, 2021-10-06:
im using rammus 93 image and yes i still get that loading in apps but reading this : https://github.com/sebanc/brunch/issues/813
i think there is some preloading which is not been able to done because even after 2 mins there was disk lighting showing in CPU case which indicates something is loading i think i need to first wait for it, when it stops then im meant to login right?

sebanc, 2021-10-06:
ChromeOS r92 introduced an increase in cases where the android container will fail to launch, I tried to improve the android_init_fix option to workaround that but it does not seem to work in your case. I will try to see if I can further improve it.

sebanc, 2021-10-06:
Otherwise you could also help to find a way to improve the android init fix. Globally I added a script which runs in a loop until the android container is supposedly ready to start (line that contains "uses 32-bit capabilities" in dmesg") in /usr/sbin/arc-init-wait.

You could try to see if it works better by adding an additional sleep delay or to tweak it some other way if you have ideas.

sebanc, 2021-10-06:
Regarding options, the config is reset every time you run the setup menu, just run the setup menu again and don't tick android_init_fix if you don't want it.

poly3dfree, 2021-10-06:
Thanks so much sebanc.
I finally ran config setup and did everything default (no fix no hyperthreading) and i waited on login screen till disk indicator stopped then logged in and waited again till it stopped then i run play store and im finally able to run apps. But when I wait only on login screen and open play store without the loading is being done it doesnt work again.

So basically who has same problem should try waiting on both login and after login to finish disk loading.
Again thanks so much sebanc for this wonderful project.

closed 2026-02-10 by github-actions[bot]