sebanc/brunch #1280: Crostini does not work: Error starting penguin container: 5 Launching vmshell failed: Error starting crostini for terminal: 5

JoseDeJesusDelaCruzTorres10, 2021-10-30:
I wanted to use Linux (Crostini) in Brunch, however this will not start, since I read that there is a detail with the microcodes from Chrome OS 91, and I don't know if there is any way to run Crostini with an Intel pentium 2020M from 2012, (which has Intel's Virtualization technology), if there is any way to modify the Grub or something like that to start Crostini, I will thank you in advance.

sebanc, 2021-10-31:
Maybe try disabling crostini gpu acceleration in chrome://flags

dorp007, 2021-10-31:
Hello Sebanc,

I hope you are doing well! May I please have a list of avelible crosh terminal commands and the man pages?

Thank you in advance.

Sincerely,

jjdct, 2022-04-23:
It was fixed with a Chrome OS update.

closed 2022-04-23 by jjdct