sebanc/brunch #1281: The sound does not work and the cooling fan is a problem-Acer chromebook spin13

TyanGithub, 2021-10-31:
Hello, first of all, thanks to the developers of these efforts, my machine is running a double boot, brunch is using the native chrome os, the current audio device is not recognized and fan hot fans do not change speed, if added "native_chromebook_image" option will cause The Android container can not open, audio devices can recognize but no sound problems, has been tested under all kernelsHello, first of all, thanks to the developers of these efforts, my machine is running a double boot, brunch is using the native chrome os, the current audio device is not recognized and fan hot fans do not change speed, if added "native_chromebook_image" option will cause The Android container can not open, audio devices can recognize but no sound problems, has been tested under all kernels.
[cb5.4-dmesg.txt](https://github.com/sebanc/brunch/files/7448746/cb5.4-dmesg.txt)

sebanc, 2021-10-31:
Hello,

If I am not mistaken, nami is the official recovery image for this device, could you confirm that this is the recovery image you have used with brunch ?

Is nami still on 4.4 kernel on official chromeos ?

If so, the best combination should be nami recovery + kernel-chromebook-4.4 + "native_chromebook_image" option.

TyanGithub, 2021-11-01:
> Hello,
>
> If I am not mistaken, nami is the official recovery image for this device, could you confirm that this is the recovery image you have used with brunch ?
>
> Is nami still on 4.4 kernel on [...]

Thank you for your reply. Yes, I use the official recovery image of Nami and have tried to use it on brunch-r90 - r93. Every kernel option has been tried, and the problem remains

TyanGithub, 2021-11-01:
This is a log running under kernel 4.4 "native_chromebook_image"（ps：Once I use this option, the Android container cannot be opened, and the sound card can be recognized but still has no sound.
[dmesg.txt](https://github.com/sebanc/brunch/files/7450610/dmesg.txt)
）

sebanc, 2021-11-01:
Have you tried kernel-chromebook-4.4 without "native_chromebook_image" option ?

sebanc, 2021-11-01:
If sound does not work, what is the output of "cras" command with kernel-chromebook-4.4 and without "native_chromebook_image" ?

TyanGithub, 2021-11-01:
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/utils.c:261:(uc_mgr_config_load) could not open configuration file /usr/share/alsa/ucm2/HDA Intel PCH/HDA Intel PCH.conf
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/parser.c:1605:(load_master_config) error: could not parse configuration for card HDA Intel PCH
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/main.c:963:(snd_use_case_mgr_open) error: failed to import HDA Intel PCH use case configuration -2

TyanGithub, 2021-11-01:
the output of "cras" command with kernel-chromebook-4.4 and without "native_chromebook_image" option

sebanc, 2021-11-01:
Sorry, forgot to ask you to enable "disable_intel_hda" option...

Could you post the output of "cras" after enabling it ?

TyanGithub, 2021-11-01:
1. with ”native_chromebook_image“ option: the fan can be adjusted automatically. The sound card can be recognized (as the volume adjustment panel can slide and the speakers can be seen), but there is still no sound. Meanwhile, the Android container cannot be opened.

2.  without "native_chromebook_image" option: the sound card is not recognized. The fan will not adjust automatically. But the Android container works.
3. I am currently using  without "native_chromebook_image" option.

TyanGithub, 2021-11-01:
[dmesg.txt](https://github.com/sebanc/brunch/files/7453584/dmesg.txt)

> Sorry, forgot to ask you to enable "disable_intel_hda" option...
> 
> Could you post the output of "cras" after enabling it and actually post another dmesg ?

TyanGithub, 2021-11-01:
chronos@localhost / $ cras
chronos@localhost / $ 

It shows nothing

sebanc, 2021-11-01:
Could you post the same thing without "snd-intel-dspcfg.dsp_driver=1" on the kernel command line ?

TyanGithub, 2021-11-01:
> Could you post the same thing without "snd-intel-dspcfg.dsp_driver=1" on the kernel command line ?

chronos@localhost / $ cras
chronos@localhost / $ 

still nothing
[dmesg.txt](https://github.com/sebanc/brunch/files/7453787/dmesg.txt)

sebanc, 2021-11-01:
Could you try this new build with kernel-chromebook-4.4 and "native_chromebook_image" option ?
https://github.com/sebanc/brunch-unstable/releases/tag/r94-unstable-20211101

TyanGithub, 2021-11-01:
> Could you try this new build with kernel-chromebook-4.4 and "native_chromebook_image" option ? https://github.com/sebanc/brunch-unstable/releases/tag/r94-unstable-20211101

Thank you for your work! The good news is that the Android container is up and running under these conditions, but there is still a problem with "headphone,speakers can be recognized but no sound."

after "cras" 
chronos@localhost / $ cras
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/utils.c:261:(uc_mgr_config_load) could not open configuration file /usr/share/alsa/ucm2/kblda7219max/kblda7219max.conf
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/parser.c:1605:(load_master_config) error: could not parse configuration for card kblda7219max
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/main.c:963:(snd_use_case_mgr_open) error: failed to import kblda7219max use case configuration -2
chronos@localhost / $ 

dmesg:
[dmesg.txt](https://github.com/sebanc/brunch/files/7454691/dmesg.txt)

sebanc, 2021-11-01:
Could you run the below command in crosh shell, reboot and try sound:
`sudo cp -r /usr/share/alsa/ucm/kblda7219max.akali360 /usr/share/alsa/ucm/kblda7219max`

TyanGithub, 2021-11-01:
> Could you run the below command in crosh shell, reboot and try sound: `sudo cp -r /usr/share/alsa/ucm/kblda7219max.akali360 /usr/share/alsa/ucm/kblda7219max`

thx but still not work 

cras:

chronos@localhost / $ cras
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/utils.c:261:(uc_mgr_config_load) could not open configuration file /usr/share/alsa/ucm2/kblda7219max/kblda7219max.conf
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/parser.c:1605:(load_master_config) error: could not parse configuration for card kblda7219max
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/main.c:963:(snd_use_case_mgr_open) error: failed to import kblda7219max use case configuration -2
chronos@localhost / $

sebanc, 2021-11-01:
Try:
`sudo mv -r /usr/share/alsa/ucm/kblda7219max/kblda7219max.akali360.conf /usr/share/alsa/ucm/kblda7219max/kblda7219max.conf`

reboot and try sound

TyanGithub, 2021-11-02:
> Try: `sudo mv /usr/share/alsa/ucm/kblda7219max/kblda7219max.akali360.conf /usr/share/alsa/ucm/kblda7219max/kblda7219max.conf`
> 
> reboot and try sound

cras

chronos@localhost / $ cras
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/main.c:411:(execute_sequence) unable to execute cset 'name='Pin 5 Mux' 1'
ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/main.c:1733:(set_verb_user) error: failed to initialize new use case: HiFi
chronos@localhost / $ 

chronos@localhost / $ grep "" /sys/class/dmi/id/* 2> /dev/null
/sys/class/dmi/id/bios_date:06/04/2020
/sys/class/dmi/id/bios_vendor:coreboot
/sys/class/dmi/id/bios_version:MrChromebox-4.12
/sys/class/dmi/id/board_asset_tag:
/sys/class/dmi/id/board_name:Akali 360
/sys/class/dmi/id/board_vendor:Google
/sys/class/dmi/id/board_version:1.0
/sys/class/dmi/id/chassis_asset_tag:
/sys/class/dmi/id/chassis_type:3
/sys/class/dmi/id/chassis_vendor:Google
/sys/class/dmi/id/chassis_version:
/sys/class/dmi/id/modalias:dmi:bvncoreboot:bvrMrChromebox-4.12:bd06/04/2020:svnGoogle:pnAkali360:pvr1.0:rvnGoogle:rnAkali360:rvr1.0:cvnGoogle:ct3:cvr:
/sys/class/dmi/id/product_family:Google_Nami
/sys/class/dmi/id/product_name:Brunch
/sys/class/dmi/id/product_sku:sku11239
/sys/class/dmi/id/product_version:1.0
/sys/class/dmi/id/sys_vendor:Google
/sys/class/dmi/id/uevent:MODALIAS=dmi:bvncoreboot:bvrMrChromebox-4.12:bd06/04/2020:svnGoogle:pnAkali360:pvr1.0:rvnGoogle:rnAkali360:rvr1.0:cvnGoogle:ct3:cvr:
chronos@localhost / $

TyanGithub, 2021-11-02:
> cras
>
> chronos@localhost / $ cras ALSA lib /build/nami/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/main.c:411:(execute_sequence) unable to execute cset 'name='Pin 5 Mux' [...]

still not work  TAT

sebanc, 2021-11-02:
The only thing that would make sense for this is that NAMI does not use 4.4 kernel anymore. I would guess it has been updated to 5.4, it is just a guess though and it would be good to confirm this.

TyanGithub, 2021-11-02:
> The only thing that would make sense for this is that NAMI does not use 4.4 kernel anymore. I would guess it has been updated to 5.4, it is just a guess though and it would be good to confirm this.

Thank you very much！It worked. It turned out that I forgot to switch the kernel back to 4.4. I used to use the 5.4 kernel.

sebanc, 2021-11-03:
Could you try the below build ?
https://github.com/sebanc/brunch-unstable/releases/tag/r94-unstable-20211103

Normally sound should work with kernel-chromebook-4.4 and "native_chromebook_image" without needing additional commands.

TyanGithub, 2021-11-03:
> Could you try the below build ? https://github.com/sebanc/brunch-unstable/releases/tag/r94-unstable-20211103
> 
> Normally sound should work with kernel-chromebook-4.4 and "native_chromebook_image" without needing additional commands.

Well, after I update this version in the shell, I can really use the speaker directly. Since I have used the command line before, I'm not sure whether it still needs to enter the command line,

sebanc, 2021-11-03:
With this version, you should not need to enter any command line to have working sound.

Just to confirm everything is ok, could post the output of the below command in crosh shell ?
`ls -l /usr/share/alsa/ucm/kblda7219max/`

TyanGithub, 2021-11-03:
> ls -l /usr/share/alsa/ucm/kblda7219max/

chronos@localhost / $ ls -l /usr/share/alsa/ucm/kblda7219max/
total 16
-rw-r--r--. 1 root root 11772 Nov  3 14:15 HiFi.conf
-rw-r--r--. 1 root root    97 Nov  3 14:15 kblda7219max.conf
chronos@localhost / $

sebanc, 2021-11-03:
It is exactly what it should look like, I guess this issue is fixed.