sebanc/brunch #1282: how can I turn on ssh into the chromebook running brunch?

cmsanta, 2021-10-31:
Hi: 

How can I enable ssh login into my chromebook running brunch from another computer running Linux? For example, I am not interested in running ssh from Crostini, I just want to able to ssh from the cros shell environment into another computer running linux.

The cros_debug is already included in the kernel command line and also I am able to enable, and the wiki doesn't mention any of this.

As a pointer, I am able to ssh from within brunch to the outside world but not the other way around, why?

cmsanta, 2021-11-01:
Issue is now resolved after trying option #3 from here: https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/chromiumos-overlay/chromeos-base/chromeos-sshd-init/files/openssh-server.conf.README

closed 2021-11-01 by cmsanta

dorp007, 2021-11-01:
https://ci.chromium.org/p/chromium/g/main/console