sebanc/brunch #1283: Add 5.10 OEM Kernel to Brunch

DavidIlie, 2021-11-01:
Hi,

Is there a way to re-compile Brunch to use the OEM version of the 5.10 kernel? I think this might be a way to resolve the problems that I have been having with my Dell XPS 13 9310 (see issue [here](https://github.com/sebanc/brunch/issues/1239)).

I think that this would solve the problem as I have the same graphical problems with Pop! which uses the same Linux kernel as brunch (5.10). However, if I use Ubuntu, I am using kernel 5.10-OEM which doesn't have any graphical problems and everything works perfectly, which might bring chance for it to work in ChromeOS as well.

Is there a way to just insert the kernel there and rebuild Brunch?

DavidIlie, 2021-11-01:
I did that and the problem is still happening, there are still graphical glitches when I change tabs, etc.

sebanc, 2021-11-01:
Try this build with the kernel 5.10 and "oled_display" option (disregard the option name, I added the patch there as I did not want to create issues for other users):
https://github.com/sebanc/brunch-unstable/releases/tag/r94-unstable-20211101

DavidIlie, 2021-11-01:
Alright, trying it now with these options:

```options=enable_updates,internal_mic_fix,no_camera_config,pwa,i915.enable_psr=0,oled_display```

closed 2021-11-09 by DavidIlie