sebanc/brunch #1285: touch screen not working after login

antoninchadima, 2021-11-02:
Hi, thank you very much for your work Sebanc. For me is Chrome OS the best Linux distribution available. And because I was not satisfied with the Chromebooks on the market, I knowingly bought an ASUS ZenBook UX393E to use with Brunch. It is very stable, and I use this computer daily for all my work, entertainment, and gaming. I would not swap for a proper Chromebook.
Most things are on the ZenBook working out of the box. Only the touchscreen has a funny issue. I do not need the touchscreen necessarily - but it is still weird. The touchscreen is working on the login page, and after logging in, it stops working. Is there a solution to this problem?

sebanc, 2021-11-03:
I think you can disable the touchscreen with CTRL+ALT+SHIFT+T, could you try it again to verify that you have not disabled the touchscreen with this key combination by mistake ?

antoninchadima, 2021-11-03:
thanks for navigating in the right direction
CTRL+ALT+SHIFT+T is actually for switching between tablet and desktop mod (no touchscreen enabling disabling)
you have to chrome://flags/#ash-debug-shortcuts Enable and use the Search + Shift + T to disable/enable the touchscreen

now it is working

closed 2021-11-03 by antoninchadima

347H3W, 2022-09-16:
Hi, for the past few months, I have been facing the same issue with my Dell Inspiron 13 5000 Series. I have been using the Brunch version of ChromeOS for years now and haven't come across this issue. Now, everything works fine except the touchscreen. I mostly use the touchscreen while using my laptop so this issue has hit me pretty hard. The touchscreen works while on the login page, and after logging in, it stops working. Please look into this issue and help me find a solution.