sebanc/brunch #1286: Chrome os not booting even if I have UEFI bios.

sajidshahriar72543, 2021-11-02:
Hello,

I am trying to install Chrome OS rammus with the help of Linux Mint to my old Lenovo Ideapad 100s laptop.

It installs without any error and  says CHROME OS INSTALLED at the end. But when I reboot and unplug usb, laptop shows no os installed and returns back to boot menu. I tried this with other linux distros, every distro results in same error.

As my hdd is mmcblk1 so I changed installation directory of my install script accordingly. Is there anything else I'm missing out?

Thanks.

The-Chosen-Sausage, 2021-11-02:
are you doing a dual boot installation?
if so, you need to install grub on your hdd first and add chromeos boot entry manually.

otherwise, your laptop may not support EFI boot or it have been disabled in bios.
try enabling EFI boot in bios or take a look at "Limited support for MBR/BIOS devices" section in Readme.md

sajidshahriar72543, 2021-11-02:
> are you doing a dual boot installation? if so, you need to install grub on your hdd first and add chromeos boot entry manually.
>
> otherwise, your laptop may not support EFI boot or it have been [...]

Thanks for the reply. 

First of all I am not installing it along side with any os. Thus it is not a dual boot. In youtube I saw a guide where the guy used linux mint as the base and installed chrome os. I did the same. According to them, the os should boot after it is installed correctly. But for me it says no os installed. And yes GRUB is not getting installed with chrome os. That maybe the case of not booting into chrome os.

Also my device is UEFI native and it is GPT partitioned device.

Thanks.

The-Chosen-Sausage, 2021-11-02:
are you sure your hdd is mmcblk1?
mmcblk usaully stands for emmc devices (SD card, USB etc..).
an hdd should be /dev/sda1 or something...

sajidshahriar72543, 2021-11-02:
> are you sure your hdd is mmcblk1? mmcblk usaully stands for emmc devices (SD card, USB etc..). an hdd should be /dev/sda1 or something...

Yes. My HDD is only 32gb in Lenovo Ideapad 100s. It uses a 32GB storage device which is soldered on the motherboard. Also GParted recognizes it as "/dev/mmcblk1 (29.12GiB)"

Thanks.

The-Chosen-Sausage, 2021-11-02:
Sounds like your pc is not recognizing the efi system partition. 
Brunch formats the efi part as fat16, you can try to re-formatting it as fat32.( copy everything inside the efi partition, formats it as fat32, and move everything back. You can find the efi partition with Linux mint's Disk program)

I'm not sure if that will solve the problem...

The-Chosen-Sausage, 2021-11-02:
If that didn't work, you can try to clear the entire disk and install grub first:
1.Copy everything from brunch's efi part to a safe place.
2.delete all partitions and create a 200mb fat32 partition at the beginning of the disk, and copy brunch's efi files in it. 
4.Reboot your computer and see if  the grub menu loaded correctly.
If so, you can follow the dual boot installation steps to install chromeos.

riglaze, 2021-11-25:
I faced the same with Rammus and all I had to do was to create a new boot entry in the UEFI Menu.

So, try this.
0.) See if ChromeOS boot entry is present in the UEFI Menu.
1a) If not, add a boot entry for ChromeOS from the UEFI menu. (I referred an article that I lost, ouch. It's basically going to UEFI and adding a boot entry and naming it and leaving the pci blablabla option as is and the path of efi file would be in EFI partition. Do Google it and lmk if you found it :))
1b) If it exists, try deleting it and re-adding it. Or follow Sausage's idea.

If that doesn't help, welp, idk.
Hopefully it does 😉

DhairyaDeveloper, 2022-02-13:
pls help me

userosos, 2022-02-21:
Hi! I create the img file uses the script after it i write the img files on an sd-card into my tablet chuwi hi10 go (Celeron n5100). Then i rebooted the tablet and i cant see in booting menu the sd card with chromeOS.

ItslilBabich, 2022-05-21:
I have the same problem, I tried installing Samus and rumms, both of them installed but when I tried it said no os found or _ and i tried with USB and Linux mint and still doesn't work

DhairyaDeveloper, 2022-05-22:
Ya in my old pc there was the same problem
There's no problem in Chrome OS
The problem is in Your Processor If your processer like Mine Is Intel Core
i3 4005U
U means Ultra Low Voltage/Power So that must be the problem.

Regards,
DhairyaDeveloper

DhairyaDeveloper, 2022-05-22:
If ur processor don't have U then just email me back I will give u some
tutorials about how to install Chrome OS.

ItslilBabich, 2022-05-22:
I don't have a U in my CPU but I have an I3 4th gen, I also installed chrome os flex to see if I have the same problem. I didn't but it did crash a lot

riglaze, 2022-05-22:
All you guys had to do was to create a boot entry in your UEFI settings. Nothing to do with the processor. 

The thing is, older UEFIs are not able to create a boot entry for ChromeOS post installation.

https://www.thomas-krenn.com/en/wiki/Restoring_UEFI_boot_entry_via_motherboard_replacement_or_BIOS_update

Perform the "Entering the UEFI Boot Entry manually in BIOS" step and you should be good to go :)

ItslilBabich, 2022-05-22:
I will try but I did open the USB and Linux with the UEFI

closed 2026-02-09 by github-actions[bot]