sebanc/brunch #1293: Keyboard is Not Working

bl4ckck, 2021-11-07:
Hello, I'm using Brunch r93 stable 20211002 and Rammus 94 (14150.64.0). 

My laptop can't type any words, but the functions key like volume up, down and brightness it's working. I already add this kernel command "module_blacklist=intel_hid ", but still not working.

====================================================================
System Manufacturer	Hewlett-Packard	
System Model	HP Pavilion 11 x360 PC	
System Type	x64-based PC
Processor	Intel(R) Celeron(R) CPU  N2840  @ 2.16GHz, 2159 Mhz, 2 Core(s), 2 Logical Processor(s)

sebanc, 2021-11-07:
Could you try adding "module_blacklist=intel_vbtn" to the kernel command line ?

bl4ckck, 2021-11-07:
> Could you try adding "module_blacklist=intel_vbtn" to the kernel command line ?

Thanks, I just use the same command from this https://askubuntu.com/questions/1274890/ubuntu-20-04-keyboard-not-working

closed 2021-11-07 by bl4ckck