sebanc/brunch #1294: Samsung Galaxy Book 12 inch: No cam, wifi, sound or brightness control

quochuy-94, 2021-11-07:
Hi,
First of all thank you so much for this incredible work. It has completely transform my old 2-in-1 into something of a joy to use.
Some issues that I hope you could help with
- No wifi on any kernel other than 4.19. 
- Sound card is detected and sound works via Bluetooth speaker but not internal one
- Camera doesn't work
- Brightness control doesn't work

This is the component of the device: https://www.driveridentifier.com/scan/samsung-galaxy-book-12-lte-driver/desktop/AE512B24EF7440AFBADC0C3B76C1F896

I tried rammus & nautilus recovery images, as well as all the options available in the settings (no camera fix, oled screen,..).

I'm on brunch 93 and chrome OS 93.

sebanc, 2021-11-07:
Honestly, I hesitated to buy this device a few years ago but decided not to after seeing the linux compatibility reports.

I don't see any success story with brightness on linux with this device, only xrandr software brightness workarounds which will not work on ChromeOS.
I am not sure if the camera works in linux but in any case it would not work in ChromeOS as it does not support i2c cameras.

I think for sound you will need the kernel 5.10, I will try to fix wifi in kernel 5.10.

quochuy-94, 2021-11-07:
thank you for such a quick reply. The brightness issue can be fixed with a screen filter app from google play. Not perfect but it works for me.
I don't use the camera much anyway so that's something I can live with. 
If the sound can be fixed it'll work very well as a daily driver. Looking forward to that!

sebanc, 2021-11-07:
Try kernel 5.10 with this build:
https://github.com/sebanc/brunch-unstable/releases/tag/r94-unstable-20211107

quochuy-94, 2021-11-07:
Wifi works with kernel 5.10 & r94 unstable. Still no sound from the internal speaker though. 
It does recognize the sound card, however no sound coming out.

sebanc, 2021-11-07:
Could you post a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" and attach dmesg.txt from your Downloads folder)

Then, boot with "module_blacklist=snd_hda_intel" added on the kernel command line, test sound and post another dmesg.