sebanc/brunch #1303: can't start vmc start termina .

silviuXgaming, 2021-11-10:
hi all have a proablem after destroy termina...cant start termina ( vmc termina start ) this is error -crosh> vmc start termina 
Error: operation vm_start failed: bad VM status: VM_STATUS_FAILURE: Failed to get image paths: termina-dlc was not installed, its state is: 0
[ERROR:src/main.rs:181] ERROR: command failed - tahsnks all

ycavan, 2021-11-11:
I fought this problem just a bit ago & how I fixed it was to add "enable_updates" option and update chromeos.  Then I had to enable #crostini-use-dlc.  After that, I could start termina from cros>

silviuXgaming, 2021-11-13:
thanks for your help , i just change it on w11 , i found more bugs , thanks

SayantanRC, 2022-04-19:
@ycavan thanks a lot. However for me with the flag #crostini-use-dlc the setup got stuck during the downloading stage. It happened thrice. Then I just disabled the flag, kept "enable_updates" and it restarted. Now it works.
Chrome OS 100. Brunch 100.