sebanc/brunch #1311: ASUS G513IH-HN086T Laptop trackpad, keyboard, wifi, bluetooth not working properly

Zandacrus, 2021-11-14:
Hello! I want to thank all of you  for your incredible work on this brunch project to bring chrome os to users like us ! 😃
I followed every instruction given [here](https://github.com/sebanc/brunch/blob/master/install-with-windows.md) and made a bootable chrome os usb using "Brunch r93 stable 20211002" and zork image (version 93). I also chose kernel 5.10 from the brunch configuration menu. But, when I booted from that usb the chrome os was extremely unstable. Problems I encountered ---

-  When I pressed any key on my keyboard the screen blacked out, after pressing another key the screen came up again and then again on any key press it blacked out.
- Trackpad was not working.
- Neither Wifi nor the Bluetooth worked. The ethernet worked though.
- Google Play store could not be installed.

My laptop specs ---

- Ryzen 7 4800H Processor
- 8 GB DDR4 Ram
- Integrated AMD Raedon Graphics + 4GB Nvidia Graphics
- WiFi : MediaTek Wi-Fi 6 MT7921 Wireless LAN Card
- Bluetooth : Bluetooth Device (Personal Area Network),  BTH\MS_BTHPAN (Hardware ID)
- Ethernet : Realtek PCIe GbE Family Controller , PCI\VEN_10EC&DEV_8168&SUBSYS_208F1043&REV_15 (Hardware ID)

Things I tried ---

- The first thing I tried was choosing kernel 5.4 from the brunch configuration menu. But after that I got stuck on the chrome os startup logo (white background).

- After a lot of googling I found [here](https://askubuntu.com/questions/1346617/ubuntu-20-04-does-not-have-mediatek-driver-mt7921-for-wifi-bluetooth) that linux kernel 5.12 and above supports my wifi device . So I installed Ubuntu 21.10 (kernel 5.13) and everything was working fine except bluetooth. So I tried to build brunch with linux kernel 5.15.1. But I failed resulting in some errors.

    - I followed the official guide to build brunch -- https://github.com/sebanc/brunch/blob/master/BUILDING.md
    - I found the kernel from here -- https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.1.tar.gz

Please anyone help me to solve my issue !! 
I'll also help as much as I can. 😄

Also, a gentle request, can anyone please build brunch with the latest linux kernel 5.15.2 (stable) ?
This will help a lot of users who has newer asus laptops like mine.

closed 2021-11-28 by Zandacrus

mbakerdeqode, 2022-01-16:
Hi did you find any solutions or did you find any linux OS which works fine with no bugs on your ROG? I happen to have the same laptop but couldn't install any linux OS.

Zandacrus, 2022-01-17:
> Hi did you find any solutions or did you find any linux OS which works fine with no bugs on your ROG? I happen to have the same laptop but couldn't install any linux OS.

I guess the wifi, trackpad, keyboard works fine with the r94 release. Though the bluetooth still doesn't work properly. If you only want any linux os, then [Ubuntu 21.10](https://ubuntu.com/download/desktop#:~:text=our%20alternative%20downloads.-,Ubuntu%2021.10,-The%20latest%20version) works fine.