sebanc/brunch #1315: No grub text after ChromeOS installation

aqn104, 2021-11-17:
Following Sebanc installation instruction from this website below: 
https://github.com/sebanc/brunch/wiki/Detailed-installation-instructions-from-Windows

- At step 31, I should have seen some text in the terminal to be saved and copy into the grub file.  However, what is see is "echo -e ... sudo update-grub" as shown below.  Can someone who is more of an expert in Linux help to to see what's wrong?
- My computer is a Spectre x360 with an i7-5500u.  

"To boot directly from this image file, add the lines between stars to either:
- A brunch usb flashdrive grub config file (then boot from usb and choose boot from disk image in the menu),
- Or your hard disk grub install if you have one (refer to you distro's online resources).
********************************************************************************
menuentry "ChromeOS" --class "brunch" {
	rmmod tpm
	img_part=/dev/sda8
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path \
			console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
	else
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	fi
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

menuentry "ChromeOS (settings)" --class "brunch-settings" {
	rmmod tpm
	img_part=/dev/sda8
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
********************************************************************************

If you have Ubuntu or Linux Mint installed, you can create the grub config automatically by running the below command:
********************************************************************************
**echo -e '#!/bin/sh\nexec tail -n +3 $0\n\n\n\n'"$(sed '1,4d;$d' $(realpath "/root/tmpmount/chromos.img").grub.txt)" | sudo tee /etc/grub.d/99_brunch && sudo chmod 0755 /etc/grub.d/99_brunch && sudo update-grub**
********************************************************************************
Copy the above grub menu entry in grub customizer"

SupraLance, 2021-11-21:
I don't see anything wrong.  The part to be copied into grub is there, above the 'If you have ubuntu/mint installed' line.  The 'echo....sudo update-grub' part below that is to be used in the terminal only if you are working in ubuntu/mint that is installed to your hard drive.

MDHACKER555MD, 2024-04-11:
Ok

Shivanshu-oss2537u, 2026-05-26:
actually when i create image ( i created it multiple times at least 10 times ) in starting it got too much error but still it asked for dual boot or enter in ubuntu terminal when i selected dualboot it gave me a file chroemos.img.grub.txt it worked fine but after some time when i used this one computer with fresh windows 10 installed it directly shows prompt shivanshu@desktop******** and whenever i am trying a new thing is appering saying 0 byte data enterd , gpt header invalid etc