sebanc/brunch #1318: Surface 6 Pro Touchscreen not working

MrMEEE, 2021-11-19:
Maybe related to - https://github.com/sebanc/brunch/issues/1249

Running on the 4.19 kernel with the options=ipts set... touch screen is not working..

If I try to remove and reinsert the intel_ipts module, it finds the device, but then spams a lot of: "invalid output buffer payload"

Any ideas??

ninjaWolf8724, 2022-01-27:
Which recovery image are you using?

SudoSnoop, 2022-09-19:
What's working for me is to build your ChromeOS image with brunch 0.97. I'm using an RAMMUS image version 102 to create it. Then with the -ipts parameter enabled and on kernel 5.4 it works as expected.
Hadn't any luck though with newer framework versions unfortunately. I mean ChromeOS is starting fine but the touchscreen never works with versions above 0.97.

CWayman, 2022-10-06:
Same here on 105 with kernel 5.15, and latest brunch tried to use another kernel with the framework options but doesn't get past the patching screen

craigwayman, 2022-10-07:
Update: I chatted on the discord with Wisteria

Solution:  Boot into windows, hold the shift key and select "Shut down" from the start menu. Keep holding shift till its off. Then boot immediately into the ChromeOS (Settings) option and set your kernel to 4.19. (ipts driver isn't down but not required for 4.19). Afterwards, boot into ChromeOS without going to windows.

All working on logon screen with stylus support.

SudoSnoop, 2022-10-07:
Gonna have a try at that once I feel like to move to a newer version of ChromeOS, since the brunch framework seems not to be compatible with ChromeOS 105 and newer.

craigwayman, 2022-10-07:
I'm on Brunch 103, Rammus 105 and Kernel 4.19 now, all working