sebanc/brunch #1323: Win11 update deleted /chromeos.img

SupraLance, 2021-11-21:
I had a triple-boot setup with chrome/brunch on an ntfs partition after the linux partition, and windows on it's own ntfs partition before the linux partition.  An update in Windows 11 deleted the /cromeos.img and left that partition nearly empty, despite the fact it really shouldn't have touched that partition at all.  It also flagged the partition as msftdata.

Is installing ChromeOS on an ext4 partition instead the recommended way to make windows leave it alone?  Or is there a better workaround since there is no way to keep Windows from updating?

ycavan, 2021-11-25:
I installed a 2nd 1tb nvme stick in my laptop and installed ChromeOS on that.  Windoze doesn't touch it.

tejasraman, 2022-01-04:
Wait… Brunch installer converts NTFS to ext4. You should create a separate EFI partition so Windows cannot interfere with Brunch/ChromeOS

closed 2026-02-04 by github-actions[bot]