sebanc/brunch #1326: Chuwi hi10 Go bootloop on installation

DeadPo01, 2021-11-23:
Tring to install ChromeOS on this device and stuck on starting from USB Flash drive.
CPU: Celeron n4500
RAM: 6GB
ChromeOS: Volteer 94
Bruch: r94 stable 20211121
Boot from USB flash drive 64Gb Sandisk
Installed ChromeOS by official instruction from GitHub on USB drive, tryed to boot on device with 4.19, 5.4, 5.10 kernels and getting bootloop anyway.. =(
The last lines that i can see in console is abot applying patches and then reboot...
What may i try else? Or what i do wrong?

fejich, 2021-11-25:
Acer Chromebook 317 uses the same CPU

Intel Celeron N4500

Maybe try Codename: `dedede`

DeadPo01, 2021-11-25:
tried rammus, volteer and dedede =(
Rammus and dedede - stuck on audio/wifi init, volteer  - is rebooting after some time after  preparing FS. I can't catch the last lines of boot.

DeadPo01, 2022-02-11:
Managed to start ChromeOS on device.
Brunch r97, image dedede 97, kernel 5.15.

DeadPo01, 2022-06-30:
Touchscreen is ok on my device.
Audio problem is in kerenel, so we need to wait for bug fixes in kernel or for new kernel.

Naced28, 2022-07-01:
I read that the audio device (of chuwi hi 10 go) is taken into account since the kernel 5.18 in linux.. I think we just have to wait ;)