sebanc/brunch #1327: [question] getting an actual ChromeOS install into the same "hacked" state?

RJVB, 2021-11-24:
Hi,

Just a quick question!
I am about to pick up a ChromeOS tablet (Asus CT100) I ordered based on my experience with Brunch. What's the most efficient sequence of commands/actions to being it in a comparably "hacked" state as a Brunch install?

Thanks!

ycavan, 2021-11-25:
honestly? put your chromebook into developer mode... install crouton, or maybe try brioche, enjoy the "hackiness."

RJVB, 2021-11-25:
Well, it's a tablet with only 32Gb of MMC so I'm going to hold off a bit on installing Crostini or similar. Turns out it's largely unnecessary under Brunch btw, if you use crew. I've even managed to run "Ferdi" (an Electron-based app) natively (but I have to start if from a croshsh).

closed 2026-02-04 by github-actions[bot]