sebanc/brunch #1328: Writing partitions failing, ls: cannot access '/usr/sbin/rootc.img': No such file or directory

aikebakeeeeees, 2021-11-25:
been trying to install brunch for a couple days now but as i finally follow the guide properly i found when trying to execute the command 'sudo chromeos-install -dst /dev/mmcblk1'
everything is fine up until writing partition 7 where it displays
ls: cannot access '/usr/sbin/rootc.img': No such file or directory
dd: invalid number: ''
and then everything is fine up until writing partition 12 it displays

Rush-er, 2021-11-25:
I have the same problem

aikebakeeeeees, 2021-12-12:
please, someone help; it's been 16 days and i am beginning to lose all hope.

MrCadawas, 2022-01-11:
I had similar issues in December although I don't rememer the exact details.

Just tried again today.  When I was making the img file in WSL, I added the -l flag (legacy efi) this time.   This seems to have created a partition layout that works for my Dell laptop.   Hope this works for you.  BTW.  I had to turn of secure boot for the legacy efi to work.