sebanc/brunch #1333: Brunch Config not getting saved

ijinesh, 2021-11-28:
Hi,
A basic question on the kernel version used in Brunch r94 stable 20211121. And how can I change kernel version to suit my requirement.

Also, I am not able to saving changes in Brunch config (using Brunch r94 stable 20211121). It says Brunch config is saved and restarts the system. However, when it restarts changes are not reflected in the config. I am trying to enable IPTS to get touchscreen working.
I am using Brunch r94 stable 20211121 on Volteer image on Surface Pro 7.
Thanks in advance.

sebanc, 2021-11-28:
Could you post your grub config ?

ijinesh, 2021-11-28:
> Could you post your grub config ? (or press "e" when you see the grub boot option and post a photo)

[image]

ijinesh, 2021-11-28:
Here it is.

ijinesh, 2021-11-28:
[image]
Is there a way to get older edit-grub-config rather than the option-based
[image]
 one currently?

aia832003, 2021-12-07:
Having the same issue here. Doesn't seem to save with new config menu. Is there a way to drop into the older grub-config?

ijinesh, 2021-12-20:
> 

You can try saving it in the Grub2win config file by inputting your options in the code file itself. In the code, in the line ending with "debug", after "debug" add options=enable_updates,ipts

This way you can enable options you want.

sokahtemen, 2021-12-23:
Same issue here. Who do i open the Grub2win config file?

ijinesh, 2021-12-24:
> 

When you create profile for the ChromeOS in Grub2Win software, you input the customer script. In that custom script, in the line ending with "debug", after "debug" add options=enable_updates,ipts

You can refer to this video for getting to the stage of copying custom script and then put your options there.
https://www.youtube.com/watch?v=jZdrosLWzk4&t=188s

novalkomik, 2022-03-04:
> > 
> When you create profile for the ChromeOS in Grub2Win software, you input the customer script. In that custom script, in the line ending with "debug", after "debug" add options=enable_updates,ipts

Unfortunately, It doesn't work properly. The `options=` were ignored by Chrome OS. I've tried it on my laptop.

ouija, 2022-04-03:
Strange, this issue just started occurring for me tonight when trying to disable the `mount_internal_drives` option and finding that it's not actually being set anymore for some reason.    When looking at the EFI, I actually don't see the `settings.cfg` file it's referring to (in the grub config) located anywhere within the `/EFI/BOOT` folder it's saying it should be...

So my question is WHERE are these settings being loaded from, and how do I disable usage of this `mount_internal_drives` option _(as my drives are STILL mounting if/when I update the config)_

yaman3238, 2022-07-28:
Having the same issue with brunch r103 , I am trying to start the touch screen service but unfortunately the features are not been saved , although i have noticed that  the splash screen is been saved. Strange!!

Losses, 2022-08-11:
Same issue here, r103 on Surface Pro 4

Losses, 2022-08-11:
Okay, here's the solution:

1. Open Terminal with `CONTROL + ALT + T`;
2. Enter `shell` and press `ENTER` to enter a normal shell;
3. Enter `lsblk` and press `ENTER` and find the NAME of the 12nd partition of your ChromeOS, its SIZE is 32M, for me, the value is `nvme0n1p12`;
4. Enter `mkdir /tmp/efi_part` and press `ENTER`;
5. Enter `sudo mount /dev/nvme0n1p12 /tmp/efi_part` (replace the partition name to yours);
6. Enter `sudo vim /tmp/efi_part/efi/boot/settings.cfg`;
7. Find `options=""`, replace it with `options="ipts"`;
8. Find `kernel="/kernel"` replace it with `/kernel-4.19`;
9. Tap `ESC`, enter `:wq` to exit the Vim (That's the hardest step :D);
10. Enter `sudo reboot` and press `ENTER`, your device will reboot;
11. Enjoy.

yaman3238, 2022-08-12:
@Losses  i have tried with your method with all kernel version and ipts_touchscreen,ithc_touchscreen on 5.10 and 5.15 but no results

Losses, 2022-08-12:
@yaman3238 use `4.19` and `ipts`, `5.10` and `5.15` won't work.

yaman3238, 2022-08-12:
@Losses I am using a teclast x3 plus 2 in 1 Tablet , did the same thing but its not working for me

Losses, 2022-08-12:
@yaman3238 Unfortunately, `ipts` is only for Surface devices

Jank8, 2022-11-18:
I have same issue, but there are changes in config file. PWA app tells me that PWA is disabled, and updates are not working anyway. Sorry for digging out.