sebanc/brunch #1336: Cant Launch Linux apps on acer c710

mikejkent090, 2021-11-30:
Hello 

so i switched to chrome os fulltime a while ago thanks to this very project.

I needed a new laptop so i bought the acer c710 as it was cheap and upgrade able. I upgraded it to 16gbs if ram, put in a 1tb Samsung solid state and put seabios on t.

Then proceeded to boot Linux and install rammus using the latest brunch release.

Everything works good with the exception of the linux Apps i cant install Linux through the settings but cannot launch terminal.

I get the following error: Starting the virtual machine Error starting penguin container: 5
Launching vmshell failed: Error starting crostini for terminal: 5

would very much appreciate any help resolving this please,

Thank you

mikejkent090, 2021-12-01:
I have read through other similar issues and disabled crostini gpu acceleration in flags but problem remains

I-Cat, 2021-12-10:
@mikejkent090
>> odd in the read me file it says  "Virtual Machines are not supported"

tejasraman, 2022-01-03:
Ask MrChromebox. This is the wrong place

mikejkent090, 2022-01-03:
Um im not asking about the bios firmware. /I'm asking about brunch the OS I
have loaded on it, so IT IS the RIGHT place. Give me a break.

tejasraman, 2022-01-03:
No, the **BIOS** has to have virtualization. support. If the BIOS doesn't have it, lunch will give errors when loading the Crostini VM.

tejasraman, 2022-01-03:
No, the **BIOS** has to have virtualization. support. If the BIOS doesn't have it, brunch will give errors when loading the Crostini VM. If you doubt me, try installing Linux and load up VirtualBox. (or do this in the chrome os shell, get to cross and type shell - `lscpu` and check for "Virtualization")

closed 2026-02-03 by github-actions[bot]