sebanc/brunch #1340: img not found. you need to load the kernel first...

MrSilverado, 2021-12-09:
the error is like :>
error: no such device: /ChromeOS/volteer_image.img.
error: file "/ChromeOS/volteer_image.img" not found
error: disk 'loop,gpt7' not found
error: you need to load the kernel first...

plz how to fix this error ;;; i am currently using ubuntu 20.04, brunch r94, volteer_image(chromeos 93)
and followed this video https://youtu.be/jZdrosLWzk4
and results are :- [image]

the grub code is ->
img_part=/dev/sda6
img_path=/ChromeOS/volteer_image.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
	cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,gpt7)/initramfs.img

pverghese, 2021-12-11:
If your's is a efi boot, try adding 
set root=(hd0,gpt6)
If it is a mbr system
 try
set root=(hd0,6)
Add it before the img_part line and see if it boots

ijinesh, 2021-12-20:
It seems there is a mismatch in the location of the image CromeOSand/or name of the partition (sda6) where you have saved the image.

mahaveer3, 2024-01-24:
> @sebanc the error is like :> error: no such device: /ChromeOS/volteer_image.img. error: file "/ChromeOS/volteer_image.img" not found error: disk 'loop,gpt7' not found error: you need to load the kernel first...
> 
> plz how to fix this error ;;; i am currently using ubuntu 20.04, brunch r94, volteer_image(chromeos 93) and followed this video https://youtu.be/jZdrosLWzk4 and results are :- [image]
> 
> the grub code is -> img_part=/dev/sda6 img_path=/ChromeOS/volteer_image.img search --no-floppy --set=root --file $img_path loopback loop $img_path linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path initrd (loop,gpt7)/initramfs.img

I too have same error : 
error: no such device: /ChromeOS/volteer_image.img.
error: file "/ChromeOS/volteer_image.img" not found
error: disk 'loop,gpt7' not found
error: you need to load the kernel first...

My grub code is :

menuentry "Brunch settings" --class "brunch-settings" {
	img_path=/chromeos.img
	img_uuid=006cf058-b8c4-11ee-8a0b-401c83984f12
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

how to solve this error ?

closed 2026-02-03 by github-actions[bot]