sebanc/brunch #1341: Tenho um laptop Lenovo com processador celeron.

ArReinaldo, 2021-12-10:
Meu laptop, é um lenovo ideap 110s com processador celeron  N3060. Vou conseguir instalar o Chrome os nele??

celiofk, 2021-12-13:
Teoricamente, sim. Da próxima vez, escreva em inglês, por favor.

tejasraman, 2022-01-03:
Isso é possível se você acessar https://cros.tech, obter a imagem de recuperação do "rammus" e instalá-lo dessa forma. Leia o Brunch Wiki para obter instruções, acessando a guia Wiki no repositório github

Desculpe se o português não é perfeito, usei o Google Translate :)

tejasraman, 2022-01-03:
Você também pode usar a imagem "nami" em https://cros.tech para o seu celeron N3060. Além disso, escreva em inglês. Você pode usar o Google tradutor, está tudo bem

closed 2026-02-03 by github-actions[bot]