sebanc/brunch #1345: Meet fault when preparing for the USB installation

zx5641, 2021-12-18:
Hi there,
I'm following the guide on preparing for the USB installation, and have an issue at step 8 "Install Brunch".
[https://github.com/sebanc/brunch/blob/master/install-with-windows.md#install-brunch](url)
My environment is WSL2 with ubuntu 20.04, have followed all previous steps strictly.

But, when I typed 
`sudo bash chromeos-install.sh -src chromeos_14150.74.0_rammus_recovery_stable-channel_mp-v2.bin -dst chromeos.img` 
it said 
`The ChromeOS disk image has to be installed outside of the WSL VM, please specify a path such as /mnt/<drive letter>/...`

I couldn't find any useful information about this issue, can anyone help? Many Thanks!!!!

ijinesh, 2021-12-20:
You can try putting in location where chromeos.img can be saved. For eg. \mnt\c\users\xavier\downloads\chromeos.img

zx5641, 2021-12-21:
> You can try putting in location where chromeos.img can be saved. For eg. \mnt\c\users\xavier\downloads\chromeos.img

Hi ijinesh, thanks for your replying!

Do you mean that I have to change the command to, just like,
`sudo bash chromeos-install.sh -src chromeos_14150.74.0_rammus_recovery_stable-channel_mp-v2.bin -dst \mnt\E\Downloads\chromeos.img`

I tried, but still failed, the same error prompt as before. Is my command wrong? So sorry that I am not very familiar with Linux commands.

ijinesh, 2021-12-22:
> Hi ijinesh, thanks for your replying!
>
> Do you mean that I have to change the command to, just like, `sudo bash chromeos-install.sh -src chromeos_14150.74.0_rammus_recovery_stable-channel_mp-v2.bin [...]

Is location correct? Normally, it is \mnt\e\users\xxxx\downloads\
Is your download folder in the E:\ root?

I had the same error but putting the correct location for ChromeOS.img helped. Actuall, you can put any valid location and not necessarily to the Download folder. For eg, you can put in \mnt\E\

zx5641, 2021-12-22:
> Is location correct? Normally, it is \mnt\e\users\xxxx\downloads Is your download folder in the E:\ root?
>
> I had the same error but putting the correct location for ChromeOS.img helped. Actuall, you [...]

You know what? I tried and found that it is not acceptable to use any location but the root of the partition.
I tried to use location as /mnt/e/ChromeOS.img and it works
Thanks, again!

closed 2021-12-22 by zx5641