sebanc/brunch #1355: Faiks to boot. Boot partition not found.

kidfromjupiter, 2021-12-25:
Managed to get past grub boot screen by adding 
`rmmod tpm` to grub.cfg. Apparently, there seems to be a conflict with these new machines with TPM chips. But fails to boot and says `the Boot partition was not found, falling back to shell...`

On intel 11th gen with Nvidia hybrid graphics if that helps. Used brunch 94 with volteer recovery.

[image]

ycavan, 2021-12-26:
You'll probably need to disable secure boot ( can't remember if this is really necessary or not ) and change your SATA bus from RAID/RST to AHCI.  This is all in the bios.

kidfromjupiter, 2021-12-26:
Its already done. One thing that ive noticed, is that the brunch framework doesnt seem to be working properly as well. Because upon selecting chromeOS settings, the screen briefly blanks out, and in a few seconds, a brunch framework splash screen appears. Another few seconds later, the whole laptop restarts and im back in the grub menu again.

ycavan, 2021-12-26:
Definitely sounds like brunch isn't set up properly. Normally that instant reboot means it can't boot from the grub partition. 

⁣Get BlueMail for Android ​

kidfromjupiter, 2021-12-26:
This was from the latest brunch release. Would an earlier release fix things somewhat? i'll try it out

kidfromjupiter, 2021-12-26:
Still nothing. Changed my recovery image to r90 and brunch to r90. Still getting the same . But this time, the kernel logs were a bit different. This was when trying to boot to debug mode.
[image]

 Normal mode won't even work. Just shows me the brunch splash screen and reboots.

kidfromjupiter, 2021-12-26:
I realized that there is a kernel bug that doesn't mount my internal hard drive on bootup. That's where my Chromeos installation is. It's a bug that existed on 5.4 versions of the kernel. Now my fedora installation uses 5.11 and now it's fixed. I think maybe that brunch uses 5.4 kernel as default and that's why it's failing to boot

kidfromjupiter, 2021-12-26:
Tried installing it to my primary boot drive. Still the same. Looks like brunch isnt scanning the proper partitions. It shows me that brunch scans a partition called zram0 or something. but not the one that has chromeos.img

ycavan, 2021-12-26:
Double check your bios.  Make sure that your SATA mode is AHCI, the error message in your screenshot says otherwise.
Something else to try is to use the 5.10 kernel.  

What I would recommend doing is to use a brunch usb drive, it'll be slow but you can configure it.  After you have a working configuration, then you can double check to see if your SATA drive shows up properly.  If it does, then create your brunch image onto that drive after you mount it.  Brunch will also save a .txt file onto that same partition with grub information.

kidfromjupiter, 2021-12-26:
Oh finally... Your suggestion worked. I had no option of simply switching SATA modes. I had to do a bit of research and figure out that disabling VMD controller in bios disables intel RST. 

So that means I should close this issue?

closed 2021-12-27 by kidfromjupiter