sebanc/brunch #1356: Updating to CrOS r96 or any CrOS Build powerwash my install

Lainasan, 2021-12-27:
Hello, I read the changelog on brunch stable r94 update that after updating to CrOS r96 or up it will powerwash the install.

But, does this issue will get fixed if i update to the unstable brunch build (brunch r96)?

Like, after updating to brunch r96 does updating to the next stable or any other channel release of CrOS will not get powerwash?

noka0206, 2021-12-27:
No. just use with r94.

sebanc, 2021-12-31:
Unfortunately, for now, updating to brunch r96 does not fix the ChromeOS r96 powerwash issue and it is unlikely that I will be able to prevent this at all.

I am still looking into it but without results for now.

Lainasan, 2022-01-03:
Ok thanks for the answer, I will look forward to it if this get fix.

Thanks again!

closed 2022-01-03 by Lainasan