sebanc/brunch #1358: Boot error: no such device. /Chromeos.img

kidfromjupiter, 2021-12-29:
This just a warning for new users. It seems like that brunch r94 does not add the grub options `rmmod tpm`

This is a necessary parameter when booting UEFI on TPM enabled devices such as notebooks from 2020 onwards. 

Praying to sebanc to notice and fix this. Love your work btw.

sebanc, 2021-12-29:
Which linux distro (and version) do you use ? I thought this issue had been fixed recently but maybe not in LTS builds.

In any case, could you try to  install brunch with the below custom chromeos-install.sh script and confirm if the generated config allows booting without issue ?

[chromeos-install.sh.tar.gz](https://github.com/sebanc/brunch/files/7787724/chromeos-install.sh.tar.gz)

kidfromjupiter, 2021-12-29:
I'm using fedora 35. Kernel 5.15. I'll try out the script you sent me later

kidfromjupiter, 2021-12-29:
Tried out your script. It did not detect that my system has TPM. The generated boot arguments are still the same. I don't suppose it will boot as the arguments are still the same. It should also be noted that im using the CLI commands to create the image.

EDIT:
I went through your script and saw that you were checking whether /usr/lib/grub/x86_64-efi/tpm.mod exists. Apparently, on  
fedora, that path does not exist. For one, the path is /usr/lib/grub/i386-pc. And even there, there is no tpm.mod file. One way of finding whether the system supports tpm is i think to see whether they appear in the ACPI tables.

sebanc, 2021-12-29:
The grub tpm module is not a kernel module and is not related to whether your pc has tpm or not.

I am confused because /usr/lib/grub/i386-pc indicates that your laptop uses an mbr bootloader in which case grub tpm module is not present.

It would seem that you are not booting from the grub installed by fedora but one from another distro, is this correct ?

sebanc, 2021-12-29:
In any case, could you remove the check on /usr/lib/grub/x86_64-efi/tpm.mod and see if the new config works ?

kidfromjupiter, 2021-12-29:
> The grub tpm module is not a kernel module and is not related to whether your pc has tpm or not.
>
> I am confused because /usr/lib/grub/i386-pc indicates that your laptop uses an mbr bootloader in [...]

I'm pretty sure that this was a stock install. But honestly, I've distro-hopped so much that i wouldnt be surprised if you were correct. But there are seperate EFI entries in my BIOS for all my previous distros that i really havent gotten around to cleanup. so i assume that this is booting from the stock grub installation.

kidfromjupiter, 2021-12-29:
I ran the script again with the necessary changes and got the boot arguments.
However, I suppose due to the tpm module not being available in my grub, adding `insmod tpm` command to grub gives me the error `cannot load unreferenced module`. Or something like that. Removing that option allows me to successfully boot.

closed 2026-02-03 by github-actions[bot]