sebanc/brunch #1359: Black Screen Dual Boot

eudy97, 2021-12-31:
I apologize in advance if this issue has been discussed before (couldn't find anything similar after looking for a little while) or if it is not and issue at all and I'm just stupid and did something wrong. 

Followed instructions to a T but only get a black screen when select ChromeOS in GRUB. USB/External Drive and/or Full Internal Drive install works fine.

Here is a my drive of my drive structure. Windows on nvmeon1p3 and Ubuntu on nvme0n1p5. I want to use nvme0n1p6 for brunch but after running the script all there is an chromeos.img file and a chromeos.img.grub.txt file:

[image]

Here is the output of the terminal window, no errors at all:

eudy97@CBX2-Ubuntu:~/Downloads$ lsblk -e7
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
nvme0n1     259:0    0   1.8T  0 disk 
├─nvme0n1p1 259:1    0   100M  0 part /boot/efi
├─nvme0n1p2 259:2    0    16M  0 part 
├─nvme0n1p3 259:3    0   500G  0 part 
├─nvme0n1p4 259:4    0   510M  0 part 
├─nvme0n1p5 259:5    0   500G  0 part /
└─nvme0n1p6 259:6    0 862.4G  0 part 
eudy97@CBX2-Ubuntu:~/Downloads$ mkdir -p ~/tmpmount
eudy97@CBX2-Ubuntu:~/Downloads$ sudo mount /dev/nvme0n1p6 ~/tmpmount
[sudo] password for eudy97: 
eudy97@CBX2-Ubuntu:~/Downloads$ sudo bash chromeos-install.sh -src chromeos_filename.bin -dst ~/tmpmount/chromeos.img -s 850
Not enough space to create image file, available space is 804 GB. If you think that this is incorrect, verify that you have correctly mounted the destination partition or if the partition is in ext4 format that there is no reserved space (cf. https://odzangba.wordpress.com/2010/02/20/how-to-free-reserved-space-on-ext4-partitions)
eudy97@CBX2-Ubuntu:~/Downloads$ sudo bash chromeos-install.sh -src chromeos_filename.bin -dst ~/tmpmount/chromeos.img -s 804
Creating image file
Creating partition table...
14.2MiB 0:00:00 [82.1MiB/s] [================================>] 100%            
32.0MiB 0:00:00 [ 338MiB/s] [================================>] 100%            
4.00GiB 0:00:06 [ 634MiB/s] [================================>] 100%            
32.0MiB 0:00:00 [ 450MiB/s] [================================>] 100%            
4.00GiB 0:00:05 [ 746MiB/s] [================================>] 100%            
 512 B 0:00:00 [4.34KiB/s] [=================================>] 100%            
1.00GiB 0:00:01 [ 622MiB/s] [================================>] 100%            
16.0MiB 0:00:00 [70.5MiB/s] [================================>] 100%            
 512 B 0:00:00 [26.1KiB/s] [=================================>] 100%            
 512 B 0:00:00 [13.6KiB/s] [=================================>] 100%            
 512 B 0:00:00 [23.8KiB/s] [=================================>] 100%            
32.0MiB 0:00:00 [ 351MiB/s] [================================>] 100%            
The grub config needed to boot ChromeOS has been generated in the file "/home/eudy97/tmpmount//chromeos.img.grub.txt".

If you have a linux distro installed which uses grub as bootloader, open another terminal and run the below command to generate the grub config automatically:
sudo cat /etc/grub.d/40_custom /home/eudy97/tmpmount//chromeos.img.grub.txt | sudo tee /etc/grub.d/99_brunch; sudo chmod 0755 /etc/grub.d/99_brunch; sudo grub-mkconfig -o /boot/grub/grub.cfg

Otherwise, add this grub config (lines between stars) manually to another grub bootloader:

 ****************************************************************************************** 
menuentry "ChromeOS" --class "brunch" {
	search --no-floppy --set=root --file /chromeos.img
	loopback loop /chromeos.img
	source (loop,12)/efi/boot/settings.cfg
	if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug loop.max_part=16 img_uuid=9672e549-b805-488b-a142-b1c88402c69b img_path=/chromeos.img \
			console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
	else
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug loop.max_part=16 img_uuid=9672e549-b805-488b-a142-b1c88402c69b img_path=/chromeos.img
	fi
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

menuentry "ChromeOS (settings)" --class "brunch-settings" {
	search --no-floppy --set=root --file /chromeos.img
	loopback loop /chromeos.img
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug loop.max_part=16 img_uuid=9672e549-b805-488b-a142-b1c88402c69b img_path=/chromeos.img
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
 ****************************************************************************************** 

Once the above actions are completed, you can reboot your computer and start ChromeOS
eudy97@CBX2-Ubuntu:~/Downloads$ sudo cp /etc/grub.d/40_custom /etc/grub.d/99_brunch~
eudy97@CBX2-Ubuntu:~/Downloads$ sudo rm -R /etc/grub.d/99_brunch~
eudy97@CBX2-Ubuntu:~/Downloads$ sudo cp /etc/grub.d/40_custom /etc/grub.d/99_brunch
eudy97@CBX2-Ubuntu:~/Downloads$ sudo nano /etc/grub.d/99_brunch
eudy97@CBX2-Ubuntu:~/Downloads$ sudo update-grub
Sourcing file `/etc/default/grub'
Sourcing file `/etc/default/grub.d/init-select.cfg'
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-5.13.0-22-generic
Found initrd image: /boot/initrd.img-5.13.0-22-generic
Found linux image: /boot/vmlinuz-5.11.0-41-generic
Found initrd image: /boot/initrd.img-5.11.0-41-generic
Found Windows Boot Manager on /dev/nvme0n1p1@/EFI/Microsoft/Boot/bootmgfw.efi
Adding boot menu entry for UEFI Firmware Settings
done
eudy97@CBX2-Ubuntu:~/Downloads$ sudo umount ~/tmpmount
eudy97@CBX2-Ubuntu:~/Downloads$ 

This is what my 99_brunch entry looks like: 

[image]

Any help will be greatly appreciated.

sebanc, 2021-12-31:
Could you try to add "rmmod tpm" just under each line starting with "menuentry" ?

Then run "sudo update-grub" and reboot to see if it works.

eudy97, 2021-12-31:
> Could you try to add a line with "rmmod tpm" text just under each line starting with "menuentry" ?
> 
> Then run "sudo update-grub" and reboot to see if it works.

That did it for me.

Thx sebanc.

sebanc, 2021-12-31:
No pb :)

closed 2021-12-31 by eudy97