sebanc/brunch #1362: Crosh shell not opening - illegal Instruction (core dumped)

admin-elvistrujanovic, 2022-01-02:
When I try to open Crosh shell with keyboard, crosh shell constantly close itself. After that when I try run crosh shell from terminal (Ctrl + Alt + F2) I get this error:

illegal instruction (core dumped)

I have latest Brunch and ChromeOS version so - is this bug by Brunch or by Chrome OS. And if this bug by Brunch framework, can this problem be solved. My Chrome OS image is rammus type. I don't have installed anything on my device (Android, Crostini - nothing)

tejasraman, 2022-01-03:
Best I can think of is that Brunch is sending illegal(unsupported) instructions to your CPU. What exactly is your CPU model? Maybe try a different recovery image, "caroline" or "syndra" might work?

tejasraman, 2022-01-03:
Best I can think of is that Brunch is sending illegal(unsupported) instructions to your CPU. What exactly is your CPU model? Maybe try a different recovery image, "syndra" or "syndra" might work? - FYI I'm not sebanc, and it could be something else

BCDTechPals, 2022-01-03:
I meant "syndra" or "caroline".

tejasraman, 2022-01-03:
I meant "syndra" or "caroline" - just to try :)

admin-elvistrujanovic, 2022-01-03:
> Best I can think of is that Brunch is sending illegal(unsupported) instructions to your CPU. What exactly is your CPU model? Maybe try a different recovery image, "syndra" or "syndra" might work? - FYI I'm not sebanc, and it could be something else

Processor: Intel(R) Celeron(R) CPU B820 @ 1.70GHz

tejasraman, 2022-01-03:
That's Sandy Bridge. There shouldn't be problems with Sandy Bridge, but a "fairly recent" Celeron is recommended. You could try "nami" recovery image, or use "caroline" (try caroline first). It may just be that the _rammus_ image isusing unsupported instruction sets for 2013< - there is supposedly a workaround inside Brunch, may not always work

Here is the caroline image: [https://cros.tech/device/caroline](https://cros.tech/device/caroline)
and the nami one: [https://cros.tech/device/nami](https://cros.tech/device/nami)

tejasraman, 2022-01-03:
That's Sandy Bridge. There shouldn't be problems with Sandy Bridge, but a "fairly recent" Celeron is recommended. You could try "nami" recovery image, or use "caroline" (try caroline first). It may just be that old issue that the _rammus_ image is using unsupported instruction sets for 2013< - there is supposedly a workaround inside Brunch, may not always work

Here is the caroline image: [https://cros.tech/device/caroline](https://cros.tech/device/caroline)
and the nami one: [https://cros.tech/device/nami](https://cros.tech/device/nami)

tejasraman, 2022-01-03:
That's Sandy Bridge. There shouldn't be problems with Sandy Bridge, but a "fairly recent" Celeron is recommended. You could try "nami" recovery image, or use "caroline" (try caroline first). It may just be that the _rammus_ image is using unsupported instruction sets. Try these two and get back to me if they don;'t work :)
Here is the caroline image: [https://cros.tech/device/caroline](https://cros.tech/device/caroline)
and the nami one: [https://cros.tech/device/nami](https://cros.tech/device/nami)

tejasraman, 2022-01-03:
Sorry for comment-bombing, but you may have to use "samus" (which doesn't get updates anymore) which is about the same era as your Celeron if name or Caroline don't work

tejasraman, 2022-01-03:
Sorry for comment-bombing, but you might have to use "samus" (which doesn't get updates anymore) which is about the same era as your Celeron if older images like nami or caroline don't work

tejasraman, 2022-01-03:
Sorry for comment-bombing, but you might have to use "samus" or "lulu" (which don't get updates anymore). They are about 3 yrs. newer than your Celeron if newer images like nami or caroline don't work

tejasraman, 2022-01-03:
Sorry for comment-bombing, but you might have to use "samus" or "lulu" (which don't get updates anymore) if caroline or nami don't work. They are about 3 yrs. newer than your Celeron.

tejasraman, 2022-01-04:
FYI, the Brunch page says that Baytrail or later is needed - you have the older "Sandy Bridge". You could try a different recovery ("nami" or "caroline") and if they work, you could try "[samus](https://cros-updates-serving.appspot.com//#samus)" (if you're OK with a release of Chrome OS from June 2021) because that supported your CPU.

tejasraman, 2022-01-04:
FYI, the Brunch page says that Baytrail or later is needed - you have the older "Sandy Bridge". You could try a different recovery ("nami" or "caroline") and if they don't work, you could try "samus" (if you're OK with a release of Chrome OS from June 2021) because that supported your CPU.

tejasraman, 2022-01-04:
FYI, the Brunch page says that Baytrail or later is needed - you have the older "Sandy Bridge". You could try a different recovery ("nami" or "caroline") and if they work, you could try "samus" (if you're OK with a release of Chrome OS from June 2021) - that has a higher chance, but is now end of life.

admin-elvistrujanovic, 2022-01-04:
> FYI, the Brunch page says that Baytrail or later is needed - you have the older "Sandy Bridge". You could try a different recovery ("nami" or "caroline") and if they work, you could try "samus" (if [...]
Is "nami" image supported by Google and my CPU?

tejasraman, 2022-01-04:
Yes, the “nami” image is supported until 2025 (rammus hits EOL in 2026 - a year after nami) but nami should have better support for your CPU. This is an honestly weird error, and usually, these weird errors on seemingly supported CPUs (this is the actual Sandy Bridge architecture) can be solved using a different recovery.

tejasraman, 2022-01-04:
FYI, the Brunch page says that Baytrail or later is needed - you have the older "Sandy Bridge". You could try a different recovery ("nami" or "caroline") and if they **don’t** work, you could try "samus" (if you're OK with a release of Chrome OS from June 2021) - that has a higher chance, but is now end of life - _sorry for the confusion with this comment_

admin-elvistrujanovic, 2022-01-04:
> Yes, the “nami” image is supported until 2025 (rammus hits EOL in 2026 - a year after nami) but nami should have better support for your CPU. This is an honestly weird error, and usually, these [...]

On stable and beta version, crosh problem is resolved but in dev - the problem is the same. Maybe, if I change the kernel in the brunch settings for booting this image - i can use crosh on the dev version too.

tejasraman, 2022-01-04:
Try kernel 4.19 - your CPU is more likely to have better support

tejasraman, 2022-01-04:
Ohhhh, you’re on dev builds. Do you have the Brunch PWA installed?

tejasraman, 2022-01-04:
fyi I am asking just because for dev, you need brunch-unstable installed

tejasraman, 2022-01-04:
Ohhhh… you are on ChromeOS Dev builds. You usually need brunch-unstable to use dev builds (it contains the latest Brunch code). Do you have the Brunch PWA installed on your Brunch laptop?

tejasraman, 2022-01-04:
That’s how you change from brunch-stable to brunch-unstable(brunch dev releases)

admin-elvistrujanovic, 2022-01-05:
> Ohhhh… you are on ChromeOS Dev builds. You usually need brunch-unstable to use dev builds (it contains the latest Brunch code). Do you have the Brunch PWA installed on your Brunch laptop?

I'm using Brunch Unstable

tejasraman, 2022-01-06:
Maybe a messed up grub config. Press E on the Brunch option and see what is going on. Tell me what shows up in the grub editor.

tejasraman, 2022-01-06:
FYI. I still think trying nami would fix it. Yours sounds like unrecognized hardware/CPU errors.

admin-elvistrujanovic, 2022-01-07:
> Maybe a messed up grub config. Press E on the Brunch option and see what is going on. Tell me what shows up in the grub editor.

Fixed. When I changed brunch kernel from 5.15 to 5.10, it's working now normally

tejasraman, 2022-01-07:
Great thanks!

ganeshambalakkara, 2022-05-21:
I'm too having the same problem crosh closes as soon as i press Ctrl+Alt+T.
After that when I try run crosh shell from terminal (Ctrl + Alt + F2) I get this error:
 illegal instruction (core dumped)

ganeshambalakkara, 2022-05-21:
@admin-elvistrujanovic Can please help me how u solved this issue

tejasraman, 2022-05-21:
Don't use kernel 5.15

ganeshambalakkara, 2022-05-23:
> 

I'm using kernel 5.4 only still

ganeshambalakkara, 2022-05-23:
@admin-elvistrujanovic I'm using kernel 5.4 only still

tejasraman, 2022-05-23:
What is your computer?

ganeshambalakkara, 2022-05-25:
> What is your computer?
> […](#)
Mine is hp elitebook 8460p 
320GB hardisk 4gb DDR2 ram with integrated graphics intel i5

tejasraman, 2022-05-25:
DDR2?
This laptop should work with Rammus, and it seems that your hardware is similar to of @admin-elvistrujanovic.

FYI I am NOT @admin-elvistrujanovic.

tejasraman, 2022-05-28:
Warning: Don't update to r101 on your model
Try kernel 5.10

ganeshambalakkara, 2022-05-28:
@tejasraman would i be able to update to another version?

tejasraman, 2022-05-28:
Not until @sebanc fixes the Mesa driver issue.
Try kernel 5.10 in Brunch config, **do not upgrade to r101**

soupCodez, 2022-08-17:
I have this same issue

tejasraman, 2022-08-17:
Hello @soupCodez !
We need some more context to deal with this. Could you please tell me
- Your chrome OS version
- Your Brunch version
- Your CPU, GPU, and computer model (if it is a laptop/prebuilt PC)
- Your kernel and any other configuration options set up

(Ya, that was a bit generic, sorry - but I can't really do much until I know info)

tejasraman, 2022-08-22:
@soupCodez are you still having this issue?

dhomas1, 2022-08-23:
Hello there!

I am still having this problem. I am on Brunch r103 with the image for Rammus on ChromeOS v103. My hardware is an HP Probook 67470b that uses a Core i5 3rd gen processor. I am using Kernel 5.10, but I had the same behaviour with kernel 5.4 and 5.14. 

I have Brunch installed on my USB key using the instructions from Github (not any other video instructions). When I get to the part where I want to install it to the HDD, I cannot open crosh. 

Troubleshooting steps I've tried:
- different kernel versions
-  disabling nX in the BIOS
- trying different boot options in Brunch Configuration Menu
- tried installing to the HDD via sudo from the root shell (CTRL - ALT - F2), which allowed me to install, but I still had crosh crashing on start. Android apps also can be installed but don't work, which I assume is a side effect of crosh not working

Any help would be welcome. Thank you!

tejasraman, 2022-08-23:
Hi @dhomas1 
Seems you're very thorough in your troubleshooting (great job!) but there is something wrong (it seems).

1. Are you using bios or EFI?
2. Have you reinstalled CrOS even once?
> installing Chrome OS v104 with Brunch r103 (I didn't think this would work :) )
It works most of the time 😉
3. Android is unrelated - try Android init fix or enabling VT-D (virtualization) in BIOS setup

dhomas1, 2022-08-23:
@tejasraman Thanks for the tips!

0: I've been using a ChromeOS USB key, using the instructions here: https://github.com/sebanc/brunch/blob/master/install-with-windows.md

I'll try using Ubuntu instead.

1: I've set up the image without the MBR patch. My Boot mode is set to "UEFI Hybrid (with CSM)", though I can set it up to "UEFI Native (without CSM)" to see if it helps.

2: I've installed and reinstalled CrOS multiple times, though I've only used the Rammus image (not sure which other image would be compatible with my hardware). 

3: I've set up the Android init fix, but my apps all still seem to crash.

I said I'd disabled nX (No eXecute), but in reality it was "Data Execution Prevention" that I'd disabled. In any case, it didn't change the outcome.

I'll try the changes and report back. Thanks for the advice!

tejasraman, 2022-08-23:
Use it without the BIOS patch for better boot times and less problems. The BIOS patch IMO is nothing but trouble. CSM is ok but don't use the patch.

You can use balenaEtcher to burn Ubuntu onto a USB

tejasraman, 2022-09-06:
Hi @dhomas1 ! I can provide you with an (up to date) video tutorial if you want.

Tejas

tejasraman, 2022-09-16:
@dhomas1 have you managed to get it working?

Not-Dhanraj, 2022-10-31:
hey there @tejasraman . Can you suggest brunch type for Intel(R) Celeron(R) CPU B830 ?
Thanks.
It would be amazing if you could also include a tutorial.

tejasraman, 2022-10-31:
The best I can give you is Caroline.

I'll give you a tutorial but I need to know if you're using windows or Linux.

tejasraman, 2022-10-31:
If you see #1677 I'm not sure if your celeron will work however it might. If you could try [Caroline](https://cros.tech/device/caroline) + [brunch 107-unstable](url)https://github.com/sebanc/brunch-unstable/releases/tag/r107-unstable-20221029) that might work.

You can read the official docs. If you want singleboot I can help but dualboot is a mixed bag (especially when it comes to fedora)

Not-Dhanraj, 2022-11-01:
> 

I don't intend to ise btw line anyways lamo😂
 I use arch cause it is pretty lightweight on my b830 laptop. Tried debian as well but arch felt smooth so going with it.
Haven't updated/turned it on from months now i need to chroot and update everything 😂

Not-Dhanraj, 2022-11-01:
> If you see #1677 I'm not sure if your celeron will work however it might. If you could try [Caroline](https://cros.tech/device/caroline) + [brunch [...]

Yeah, i want singleboot. If it'll work i'll upgrade to a 150 gig ssd.
Are you sure that unstable might not be unstable? 🤔
Can it work with stable?

tejasraman, 2022-11-01:
I mean it will but unstable is never unstable. I wanted to replicate my Sandy bridge setup but stable _should_ work.

Single boot :)

(BTW I **don't** use Arch)

tejasraman, 2022-11-01:
I mean it will but unstable is never unstable. I wanted to replicate my Sandy bridge setup but stable _should_ work.

Single boot :) I can help 

(BTW I **don't** use Arch)

tejasraman, 2022-11-01:
I'll give you a tutorial in any form you want.

Not-Dhanraj, 2022-11-01:
> I'll give you a tutorial in any form you want.

Video tutorial would be amazing 😅

tejasraman, 2022-11-01:
Will do! Want me to do it on arch?

Not-Dhanraj, 2022-11-01:
> Will do! Want me to do it on arch?

Do it with whatever you have right now, or whatever you are comfortable with😁

[15 more replies]

closed 2026-02-03 by github-actions[bot]