sebanc/brunch #1370: Graphics Issues - Surface Pro 2

ZProLegend007, 2022-01-08:
### Discussed in https://github.com/sebanc/brunch/discussions/1369

<div type='discussions-op-text'>

<sup>Originally posted by **ZProLegend007** January  8, 2022</sup>
Hi There!
I'm relatively new to changing operating systems on different computers but i'm getting the hang of things. I just installed Chrome os on my Surface pro 2 and i'm having a few graphical issues, I've been using asphalt 9 to test my fixes to little or no avail. I'm not sure if it is officially supported on this laptop but if there is any way to fix this it would be greatly appreciated. I will attach some images below to compare what this loading symbol does and should look like.

[image]
[image]
</div>

ZProLegend007, 2022-01-08:
I have also realised that the GPU Drive isn't taking up any memory, could that be related?

tejasraman, 2022-01-14:
Wait. How did you find that? htop? Chrome Diagnostics?

tejasraman, 2022-01-14:
Also try kernel 5.10, maybe there was the old problem of some Intel GPUs being borderline unusable under 5.4/5.5 _due to driver issues_?

ZProLegend007, 2022-01-17:
I've tried all the kernels, I used the ash-debug-shortcuts to view my memory usage (Ctrl+Shift+Alt+G). I have switched back to Windows,

closed 2022-01-17 by ZProLegend007