sebanc/brunch #1371: Google play apps and services not working

opkpEh, 2022-01-09:
I've installed and reinstalled chrome os many times, when i installed chrome os for the first time "google play apps and services" were working, but after that they haven't. I've tried everything on internet , please help.

My laptop : HP Pavilion Gaming Ryzen 5 Quad Core 3550H
gpu : nvidia 1650
cpu : ryzen 3550H

chrome book version: 96.0.4664.111 
brunch version: r94 stable 20211121 
 
[image]
The error I'm getting. 

If there's any other info that you need feel free to reach out to me.

thanks for reading.

Just in case:
My email: harshpandeykp20@gmail.com

ZProLegend007, 2022-01-09:
Which chrome os build are you using? E.g rammus or octopus

ZProLegend007, 2022-01-13:
This happened to me briefly after a brunch update, maybe move to the unstable channel for brunch?

opkpEh, 2022-01-19:
thanks for reaching out , sorry I wasn't able to reply quicker, i'm on
brunch stable (latest one)

opkpEh, 2022-01-19:
I was ble to fix this btw I just spammed re-try button for like 30 times
and it worked idk why but it did

closed 2023-12-10 by opkpEh