sebanc/brunch #1372: The palm rejection feature does not work

TyanGithub, 2022-01-11:
**First of all, thank you to the developers for their efforts.**
I have a laptop that supports wacom stylus.
After installing brunch, the stylus and touch screen work fine.
But when my finger touches the screen as I write, **my writing will be interrupted.**
This doesn't happen on one of my chromebooks with brunch installed.
I tried many versions of the recovery image and different kernels to _get the same result_. Even fydeos is the result.
May I ask anyone who is in the same situation as me or who knows the solution?

tejasraman, 2022-01-14:
Try enabling the palm rejection flag in the chrome://flags menu

TyanGithub, 2022-01-14:
> Try enabling the palm rejection flag in the chrome://flags menu

thanks,but it doesn't work.
I turned on an option called touch suppression. TAT

tejasraman, 2022-01-17:
I have no idea past that

closed 2026-02-03 by github-actions[bot]