sebanc/brunch #1383: Error creating chromeos.img

hayrun-st, 2022-01-19:
Using Windows 10 WSL, Ubuntu image to create chromeos.img

I'm seeing lots of errors when building.  I did write the image to USB using Rufus and could not get it to boot on my Dell Latitude 5290 2-in-1.

Creating partition table...
losetup: cannot find an unused loop device
losetup: cannot find an unused loop device
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
1.00GiB 0:00:05 [ 173MiB/s] [================================>] 100%
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
ERROR: Can't open : No such file or directory
chromeos-install.sh: line 335: * 512 : syntax error: operand expected (error token is "* 512 ")
32.0MiB 0:00:00 [ 164MiB/s] [================================>] 100%
losetup: option requires an argument -- 'd'
Try 'losetup --help' for more information.
losetup: option requires an argument -- 'd'
Try 'losetup --help' for more information.
The ChromeOS disk image has been created. If you want to dual boot this disk image with Grub2Win type "dualboot"

sebanc, 2022-01-19:
Install WSL2, I will add an error message in the next release

hayrun-st, 2022-01-20:
That worked for me!

tanimpersonal, 2022-01-21:
worked for me. I have another question @sebanc . I have enabled the configuration of mount HDD on chrome os. Will it cause high CPU usage everytime i boot into chromeos?

FPSensor, 2023-05-10:
> Install WSL2, I will add an error message in the next release

i was trying to create chromeos.img using a chromebook with the linux env and getting same error, it is possible to use a chromebook to create the img or im wasting my time?

cyna76, 2023-08-06:
same issue , trying create image using chrome os flex

myoshida1192, 2024-01-04:
I had same issue.

```$(( $size * 512 ))``` is invalid syntax.

I change 2 steps.

### step 1. I added following lines.

```
local pv_size
pv_size=$(( size * 512))
```
### step 2. And change as follows
```$(( $size * 512 ))``` -> ```$pv_size```

**There are 2 points to add about step 1 and 4 points to change about step 2.**

This is chromeos-install.sh of r111 I used. Give it a try. Good luck.
```
#!/bin/bash
if ( ! test -z {,} ); then echo "Must be ran with \"bash\""; exit 1; fi
if [ -z $(which cgpt) ]; then echo "cgpt needs to be installed first"; exit 1; fi
if [ -z $(which pv) ]; then echo "pv needs to be installed first"; exit 1; fi
if [ $(whoami) != "root" ]; then echo "Please run with this script with sudo"; exit 1; fi

usage()
{
	echo ""
	echo "Brunch installer: install ChromeOS on device or create disk image from the brunch framework."
	echo "Usage: chromeos_install.sh [-s X] [-l] -src chromeos_recovery_image_path -dst destination"
	echo "-src (source), --source (source)			ChromeOS recovery image"
	echo "-dst (destination), --destination (destination)	Device (e.g. /dev/sda) or Disk image file (e.g. chromeos.img)"
	echo "-s (disk image size), --size (disk image size)	Disk image output only: final image size in GB (default=14)"
	echo "-l, --legacy_boot					Use legacy efi partition (no secure boot support)"
	echo "-h, --help					Display this menu"
}

blocksize()
{
  local path="$1"
  if [ -b "${path}" ]; then
    local dev="${path##*/}"
    local sys="/sys/block/${dev}/queue/logical_block_size"
    if [ -e "${sys}" ]; then
      cat "${sys}"
    else
      local part="${path##*/}"
      local block
      block="$(get_block_dev_from_partition_dev "${path}")"
      block="${block##*/}"
      cat "/sys/block/${block}/${part}/queue/logical_block_size"
    fi
  else
    echo 512
  fi
}

numsectors()
{
  local block_size
  local sectors
  local path="$1"

  if [ -b "${path}" ]; then
    local dev="${path##*/}"
    block_size="$(blocksize "${path}")"

    if [ -e "/sys/block/${dev}/size" ]; then
      sectors="$(cat "/sys/block/${dev}/size")"
    else
[... 449 lines cut ...]
			fi
			image_size=$1
			;;
			-l | --legacy_boot)
			legacy_boot=1
			;;
			-h | --help)
			usage
			exit 0
			;;
			*)
			echo "$1 argument is not valid"
			usage
			exit 1
		esac
		shift
	done
	if [ -z "$image_size" ]; then image_size=14; fi
fi
if [ "$type" == "Singleboot (install on a disk)" ]; then
	singleboot
elif [ "$type" == "Dualboot (create an image)" ]; then
	dualboot
fi
```

slendermon, 2024-04-21:
im on chrome os finally figured out how to do it  but needs developer mode on and package manager called chromebrew

i needed to go to tty2 and use chromebrew install pv

then it works

slendermon, 2024-04-21:
im on chrome os apparently works by tty2 but with developer mode on and package manager called chromebrew

i needed to go to tty2 and use chromebrew install pv

then it works