sebanc/brunch #1392: Stuck at "Patches are being applied"

tim103117, 2022-01-24:
The startup screen stops at "Patches are being applied", and the configuration files I use are brunch r93 and rammus recovery 93.
[image]

**I followed this method to install**
https://youtu.be/ROBpXNonVxc

**About my old PC**

Machine:   Type: Desktop Mobo: Acer model: Aspire M1935 serial: <filter> UEFI: American Megatrends
       	v: P01-A1 date: 03/12/2012
CPU:   	Topology: Dual Core model: Intel Pentium G870 bits: 64 type: MCP arch: Sandy Bridge
       	rev: 7 L2 cache: 3072 KiB
       	flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 12372
       	Speed: 3002 MHz min/max: 1600/3100 MHz Core speeds (MHz): 1: 2936 2: 3086
Graphics:  Device-1: NVIDIA GF119 [GeForce GT 620 OEM] vendor: PC Partner Limited / Sapphire
       	driver: nouveau v: kernel bus ID: 01:00.0 chip ID: 10de:1049
       	Display: x11 server: X.Org 1.20.11 driver: modesetting unloaded: fbdev,vesa
       	resolution: 1920x1080~60Hz
       	OpenGL: renderer: NVD9 v: 4.3 Mesa 21.0.3 direct render: Yes
Audio: 	Device-1: Intel 7 Series/C216 Family High Definition Audio
       	vendor: Acer Incorporated ALI driver: snd_hda_intel v: kernel bus ID: 00:1b.0
       	chip ID: 8086:1e20
       	Device-2: NVIDIA GF119 HDMI Audio vendor: PC Partner Limited / Sapphire
       	driver: snd_hda_intel v: kernel bus ID: 01:00.1 chip ID: 10de:0e08
       	Sound Server: ALSA v: k5.4.0-91-generic
Network:   Device-1: Intel 82579V Gigabit Network vendor: Acer Incorporated ALI driver: e1000e
       	v: 3.2.6-k port: f040 bus ID: 00:19.0 chip ID: 8086:1503
       	IF: enp0s25 state: up speed: 100 Mbps duplex: full mac: <filter>
Drives:	Local Storage: total: 960.95 GiB used: 6.20 GiB (0.6%)
       	ID-1: /dev/sda vendor: Seagate model: ST1000DM003-9YN162 size: 931.51 GiB
       	speed: 6.0 Gb/s serial: <filter> temp: 37 C
       	ID-2: /dev/sdb type: USB vendor: Transcend model: JetFlash Transcend 32GB
       	size: 29.44 GiB serial: <filter>
Partition: ID-1: / size: 1.90 GiB used: 27.3 MiB (1.4%) fs: overlay source: ERR-102
USB:   	Hub: 1-0:1 info: Full speed (or root) Hub ports: 2 rev: 2.0 chip ID: 1d6b:0002
       	Hub: 1-1:2 info: Intel Integrated Rate Matching Hub ports: 6 rev: 2.0
       	chip ID: 8087:0024
       	Device-1: 1-1.1:3 info: Logitech M105 Optical Mouse type: Mouse
       	driver: hid-generic,usbhid rev: 2.0 chip ID: 046d:c077
       	Device-2: 1-1.2:4 info: Transcend Information JetFlash type: Mass Storage
       	driver: usb-storage rev: 2.1 chip ID: 8564:1000
       	Device-3: 1-1.5:5 info: Realtek USB 2.0 multicard reader type: Mass Storage
       	driver: ums-realtek rev: 2.0 chip ID: 0bda:0158
       	Hub: 2-0:1 info: Full speed (or root) Hub ports: 2 rev: 2.0 chip ID: 1d6b:0002
       	Hub: 2-1:2 info: Intel Integrated Rate Matching Hub ports: 6 rev: 2.0
       	chip ID: 8087:0024
       	Device-4: 2-1.5:3 info: Logitech Keyboard K120 type: Keyboard,HID
       	driver: hid-generic,usbhid rev: 1.1 chip ID: 046d:c31c
       	Hub: 3-0:1 info: Full speed (or root) Hub ports: 4 rev: 2.0 chip ID: 1d6b:0002
       	Hub: 4-0:1 info: Full speed (or root) Hub ports: 4 rev: 3.0 chip ID: 1d6b:0003
Sensors:   System Temperatures: cpu: 29.8 C mobo: 27.8 C gpu: nouveau temp: 39 C
       	Fan Speeds (RPM): N/A

happyeggchen, 2022-01-25:
Same,after i switch to brunch 94 this can be done,i don't know why,but you can modify the boot entry in grub configs verbose to 1
to see what happened,on earth

happyeggchen, 2022-01-25:
Hey do not close this,this problem still exist,let's show it to devs

gigglypup, 2022-02-28:
i have this problem too.

ikhsanytc, 2023-09-28:
I have the same issue, and i change the partition format to ext4 and then the chrome os boot normally

Shanmukhsvstp, 2023-09-29:
I used samsung evo nvme ssd in ext4 format only. But still I unable to go forward after this step although my pc meet all requirements.

polocatfan, 2025-01-13:
imagine saying nothing then closing the issue.

polocatfan, 2026-02-01:
***hole dev