sebanc/brunch #1399: Sound not work!

the-leaf, 2022-01-29:
NEC Versa pro VT-J
CPU : Atom Z3795 Baytrail

Sound And Bluetooth not work
[Uploading dmesg_kernel_5.14.txt…]()

tejasraman, 2022-02-05:
Run `sudo lshw -c Sound` and tell me the output. Seeing lots of similar issues with Atoms recently, have not got responses from anyone I have asked to run the lshw command.

Tjis tells me what sound device you have

tejasraman, 2022-02-05:
Run `sudo lshw -c Sound` in shell(get to by pressing Ctrl+Alt+T and typing shell) and tell me the output. Seeing lots of similar issues with Atoms recently, have not got responses from anyone I have asked to run the lshw command.

This command tells me what sound device you have anyone I have asked to run the lshw command.

Tjis tells me what sound device you have

the-leaf, 2022-02-06:
> Run `sudo lshw -c Sound` in shell(get to by pressing Ctrl+Alt+T and typing shell) and tell me the output. Seeing lots of similar issues with Atoms recently, have not got responses from anyone I have [...]

[image]

tejasraman, 2022-02-08:
Try this:``  lspci -v | grep Audio ``

the-leaf, 2022-02-08:
> Try this:` lspci -v | grep Audio`

[image]
[image]

tejasraman, 2022-02-08:
Do you have a Thinkpad tablet?
Try adding broadcom_wl to the grub config for Bluetooth.
Try using kernel 5.10 in grub config for audio.

https://github.com/sebanc/brunch/blob/master/troubleshooting-and-faqs.md

tejasraman, 2022-02-08:
Sorry did not read the first post
The NEC tablet is same board as a Thinkpad Model 221b(10 203C)

the-leaf, 2022-02-08:
> Do you have a Thinkpad tablet? Try adding broadcom_wl to the grub config for Bluetooth. Try using kernel 5.10 in grub config for audio. https://github.com/sebanc/brunch/blob/master/troubleshooting-and-faqs.md
> […](#)

grub config : broadcom_wl   Bluetooth not work
kernel 5.10  = No soundcards

tejasraman, 2022-02-08:
Did you do it the way the link told you to?

truongtam-fudosan, 2022-02-09:
@the-leaf change to kernel 4.19

the-leaf, 2022-02-10:
> Did you do it the way the link told you to?

Kernel 5.10 
[image]
[image]

tejasraman, 2022-02-10:
You have RealTek audio. Try the disable_intel_hda option.

the-leaf, 2022-02-10:
> You have RealTek audio. Try the disable_intel_hda option.
> […](#)

sound not work :(
[image]

tejasraman, 2022-02-11:
This seems to be an error where the Thinkpad board like yours has an ACPI error and declares the sound card wrong. There should be a patch, but I cannot help anymore.

the-leaf, 2022-02-15:
> This seems to be an error where the Thinkpad board like yours has an ACPI error and declares the sound card wrong. There should be a patch, but I cannot help anymore.

thk. I fixed it.
sound and bluetooth this work 😀😀
[image]

tejasraman, 2022-02-16:
How?
I did hear of some patches, but…

the-leaf, 2022-02-16:
NEC Versa pro VT-J  Driver this same Lenovo Thinkpad 10 Tablet.

Fix In kernels/5.X/sound/soc/intel/common/soc-acpi-intel-byt-match.c   And kernels/5.4/sound/soc/codecs/rt5670.c

search ThinkPad 10

In  :  soc-acpi-intel-byt-match.c add

{
		.callback = byt_thinkpad10_quirk_cb,
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "NEC"),
			DMI_MATCH(DMI_PRODUCT_NAME, "PC-VK24VTAMJ"),
		},
	},

in rt5670.c add

{
		.callback = rt5670_quirk_cb,
		.ident = "Lenovo Thinkpad Tablet 10",
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "NEC"),
			DMI_MATCH(DMI_PRODUCT_NAME, "PC-VK24VTAMJ"),
		},
		.driver_data = (unsigned long *)(RT5670_DMIC_EN |
						 RT5670_DMIC1_IN2P |
						 RT5670_DEV_GPIO |
						 RT5670_JD_MODE1),
	},

tejasraman, 2022-02-16:
Really? No default patching like Linux? Looks like NEC messed up ACPI just as badly as Lenovo did.

closed 2022-09-24 by the-leaf

nh0kbin2001, 2023-01-29:
> .callback = byt_thinkpad10_quirk_cb,
> .matches = {
> DMI_MATCH(DMI_SYS_VENDOR, "NEC"),
> DMI_MATCH(DMI_PRODUCT_NAME, "PC-VK24VTAMJ")

hello, can i ask  you, how to edit 2 file?? where is these??

tejasraman, 2023-01-29:
@the-leaf built this from source. @the-leaf someone needs to make a PR for this issue. 

You need a brunch build with the patches @truongtam-fudosan

nh0kbin2001, 2023-01-30:
> patches

i.e. I have to recompile a brunch for myself.  This seems too complicated for a newbie like me.  But anyway, have to try it once.  This problem seems to originate from the linux kernel.  I found my problem the same way the other guy found it.  My machine Nec pc-vk12 it corresponds to lenovo thinkpad helix 2 https://patchwork.kernel.org/project/alsa-devel/patch/E1dAAy5-002Sr-2s@finisterre/

nh0kbin2001, 2023-01-30:
> @the-leaf built this from source. @the-leaf someone needs to make a PR for this issue.
> 
> You need a brunch build with the patches @nh0kbin2001

i.e. I have to recompile a brunch for myself. This seems too complicated for a newbie like me. But anyway, have to try it once. This problem seems to originate from the linux kernel. I found my problem the same way the other guy found it. My machine Nec pc-vk12 it corresponds to lenovo thinkpad helix 2 https://patchwork.kernel.org/project/alsa-devel/patch/E1dAAy5-002Sr-2s@finisterre/

the-leaf, 2023-03-23:
@nh0kbin2001 hi bro You can check it out at my brunch.
brunch https://github.com/the-leaf/brunch/commits/r103-VersaPro

tejasraman, 2023-03-23:
@the-leaf please submit a PR! This fix would be appreciated by many :⁠-⁠)

the-leaf, 2023-03-23:
I've created a new patch file for kernel 5.10 and a new build brunch for my own use.

Example my patch file "NEC_rt5670.patch"
`--- a/sound/soc/codecs/rt5670.c
+++ b/sound/soc/codecs/rt5670.c
@@ -2863,6 +2863,18 @@
 						 RT5670_GPIO1_IS_IRQ |
 						 RT5670_JD_MODE1),
 	},
+	{
+		.callback = rt5670_quirk_cb,
+		.ident = "NEC Versapro VT-J",
+		.matches = {
+			DMI_MATCH(DMI_SYS_VENDOR, "NEC"),
+			DMI_MATCH(DMI_PRODUCT_FAMILY, "LaVie/VersaPro"),
+		},
+		.driver_data = (unsigned long *)(RT5670_DMIC_EN |
+						 RT5670_DMIC1_IN2P |
+						 RT5670_GPIO1_IS_IRQ |
+						 RT5670_JD_MODE1),
+	},
 	{
 		.callback = rt5670_quirk_cb,
		.ident = "Lenovo Thinkpad Tablet 10",
--- a/sound/soc/intel/common/soc-acpi-intel-byt-match.c
+++ b/sound/soc/intel/common/soc-acpi-intel-byt-match.c
@@ -44,6 +44,13 @@
 			DMI_MATCH(DMI_PRODUCT_VERSION, "ThinkPad 8"),
 		},
 	},
+	{
+		.callback = byt_thinkpad10_quirk_cb,
+		.matches = {
+			DMI_MATCH(DMI_SYS_VENDOR, "NEC"),
+			DMI_MATCH(DMI_PRODUCT_FAMILY, "LaVie/VersaPro"),
+		},
+	},
 	{
 		.callback = byt_thinkpad10_quirk_cb,
		.matches = {`

tejasraman, 2023-03-23:
@the-leaf Ok, but can you add the patch file to all the kernels before submitting a PR?
Making it 5.10 specific could cause confusion and, as this applies to many devices (the thinkpad and others like your NEC) having it in more kernels will save people a great deal of trouble.

the-leaf, 2023-03-23:
> @the-leaf Ok, but can you add the patch file to all the kernels before submitting a PR? Making it 5.10 specific could cause confusion and, as this applies to many devices (the thinkpad and others like your NEC) having it in more kernels will save people a great deal of trouble.

Yes, I need to try and fix it first.
when i finished testing i will update

tomekb530, 2023-03-29:
It'll be great to add it to 6.1 kernel because in older kernels there is problem with keyboard