sebanc/brunch #1402: wifi RTL8821CE working BUT cannot connect

spvmsolo, 2022-01-31:
As the title reads, it's on my laptop, I managed to make the wifi work with networks that don't have security. But my main 5g wifi has WPA2 and it just won't connect. My router also has 2.4 but no dice. I made a shared wifi with my android and removed the security and it did connect, it works just fine but I cannot connect to anything else. I tried adding security to the android wifi but again it didn't connect.
It tries for some 15 seconds and it gives me a generic "unknown error: could not connect" message.
Windows and Kubuntu work just fine so I don't think is the antenna or anything.
Bluetooth says it's on but I have not tried to connect it with anything yet so it might have the same problem.

---------
chromeOS-volteer 94 on intel i3-10110U 11th gen 
brunch r97
hp laptop 15-dy20511a

tejasraman, 2022-02-08:
1 - Are you using 8821CE option in grub config?
2 - Brunch R97 - ChromeOS update to r98

spvmsolo, 2022-02-08:
1 - I added "options=rtl8821ce,rtbth" right after cros_debug, with a space on either side, editing grub on linux (i assume i did that right), no dice.
2 - I updated to 98.0.4758.91 using the check update button on settings, no dice.

The notification says "unknown error: not-associated" and the password window says in red "There was an error while the network was being configurated"
I tried using kernel 5.15 on first boot but it crashes right after the chrome logo shows up. I also tried not using cros_debug (on 5.10) on a first boot as well but it crashes after login.

A picture of linux nano grub edit https://imgur.com/a/A846WVB just in case

tejasraman, 2022-02-08:
FYI, you need to use the enable_updates option to get updates in Settings.
Try Brunch Unstable

spvmsolo, 2022-02-08:
I did use enable updates, that's how I updated it. I enabled it on the "chromeOS settings" thing on boot.
Will try unstable after work.

spvmsolo, 2022-02-08:
Wait, where are the unstable ones?

tejasraman, 2022-02-09:
Go to https://github.com/sebanc/brunch-unstable and download and install Brunch from this repository

or enable the pwa option in grub, install the PWA, and update to Brunch-unstable from there

spvmsolo, 2022-02-09:
updated brunch to r97 unstable 2022 0205 still no luck

tejasraman, 2022-02-09:
Try to remove the rat that in grub. Use the ChromeOS settings gs entry to edit, not Linux.

spvmsolo, 2022-02-10:
> Try to remove the rat that in grub. Use the ChromeOS settings gs entry to edit, not Linux.

Not sure why this message didn't appear here (or why I can't see it?) and I'm also not 100% sure what is means but I did edited grub using chrome with the "sudo edit-grub-config" entry. This is how I managed to make the wifi work in the first place, but I still can't connect with password protected wifis. It just ain't having it.

tejasraman, 2022-02-10:
Auto correct meant rtbth

tejasraman, 2022-02-10:
You do not need the rtbth option. Your 8821ce does not use that chipset.

tejasraman, 2022-02-10:
That rat was autocorrect, not grub :) autocorrect is a rat

spvmsolo, 2022-02-10:
In other news, while looking at the forum for answers, apparently you are supposed to have multiple partitions? I only have one with a "chromeos.img" and a "chromeos.img.grub.txt" file on it. I used the dual boot guide on here to do it and while installing I did saw a couple of partitions being created, and then nothing.
A screenshot, also I did recorded a video of my woe
https://youtu.be/0q9DnzV4Kyg
[image]

tejasraman, 2022-02-11:
Yes - there should be 12 partitions(in my experience too)

tejasraman, 2022-02-11:
Plus, Chrome OS should not be on an NTFS partition.

tejasraman, 2022-02-12:
> Today's update:
> 1.- Following a different guide I managed to run the installer as a GUI (with zenity installed). From what I can deduce: while dual booting it can only be installed as an img file; the 12 partitions come from a single boot. I will try installing on a usb pendrive first then onto the partition tomorrow.
> 2.- I tried using ext4 as the partition format, following the same guide in here for dual boot linux and...
>
>
> "chrome os was not found on any device. returning to shell"
> ntfs did worked after that.

PLEASE do NOT use someone else’s GUI installer! Please use only the ChromeOS install instructions on Brunch website.

spvmsolo, 2022-02-12:
I DID do that, sorry if I was not clear

sebanc, 2022-02-12:
The rtl8821ce module is an unofficial one which works in many cases but it is probably not 100% compatible, there is not much I can do about that.

However I wonder about the "--part-uuid" error message, which grub do you use ? Is it grub2win ?

sebanc, 2022-02-12:
Well, just in case, could you post the output of "sudo lspci" from crosh shell ?

Also, do not enable "rtbth" option

spvmsolo, 2022-02-12:
I... did not use any grub? Or I used the default one? I followed the instructions in here to Dual Boot with Linux, they're both installed on a external SDD. Linux is Kubuntu 21.10
[image]
[image]

spvmsolo, 2022-02-12:
Uh... how do I do that again? Or am I supposed to be seeing something I don't?
[image]

Ubuntu brings me to kubuntu's grub
Windows brings me to normal boot
Generic external (where kubuntu and chros are) reboots the laptop without any prompts 

Should I have the time tomorrow I will sacrifice Kubuntu and install ChrOS single boot.

tejasraman, 2022-02-12:
Boot from Efi file

spvmsolo, 2022-02-12:
Which one? there are two very long names:
The first one says:
+ ubuntu
- -  mmx64 --> shim uefi key manager
- -  shimx64  --> kubuntu
- -  grubx64  --> kubuntu

+ BOOT
- - mmx64  --> shim uefi key manager
- - fbx64  --> reboot
- - bootx64  --> reboot

The second one says:
- Microsoft
- Boot
- HP

tejasraman, 2022-02-13:
In GRUb, highlight ChromeOS, press E, and add the settings there for a test boot. To add the changes permanently, download GRUB Configurator in Kubuntu and edit ChromeOS.

tejasraman, 2022-02-13:
> I have done that before, guess what happened. Will check grub configurator later.

Guess? It didn’t work? Use GRUb configurator to change Kubuntu’s ChromeOS boot entry
I have an HP desktop with an 8821ce, should I test brunch and see if it is an issue with brunch? It is an M01-F0016

sebanc, 2022-02-13:
It's a change in brunch unstable which caused your grub issue, you need to reinstall the stable brunch. Do you know which grub version is installed ?

I just noticed in your previous comments:
```
1 - I added "options=rtl8821ce,rtbth" right after cros_debug, with a space on either side, editing grub on linux (i assume i did that right), no dice.
```

This is not how it works anymore, please use the configuration menu.

tejasraman, 2022-02-13:
I did the test on my desktop. I got the part-uuid error on unstable AND a boot loop(to add to the misery). Trying stable now, will get back

tejasraman, 2022-02-13:
Hey! I just got brunch stable to boot(no errors). I needed kernel 5.10 to boot  on my PC, but it seems that the 8821ce driver works as expected. 

[image]

tejasraman, 2022-02-13:
although my PC's AMD, so things might be different. I used the settings menu to change, so idk if there is a differnece :)

spvmsolo, 2022-02-13:
I'm so happy for you

Today's update:
1.- tejasraman says to edit grub and use the options line, sebanc says to not do that and use the configurations menu. Regardless I used both (independently and separately). Same as always: wifi works but it doesn't connect.
* For the record I did use the config the first time and this is how I got here. If I don't the wifi shows ON but the list is empty

2.- My wifi's name has special characters (í and ñ) so I thought it might have something to do with that. I used android share wifi and... it connected just fine, so it's not that.
* And as I mentioned before I can't use this 'cause android won't connect to wifi and share it at the same time. And I haven't found a jailbreak tweak for iOS 14 to do that with my ipad.
* On that note I found something weird. When I connect to share wifi it's fine, EVEN with the password. So it must be the router that's messing up, right? Well, my cousin was here today and it happens that he has a legit chromebook... and it works with my wifi, but my hackbook doesn't. So it's not that. (Lenovo Tab Ideapad duet)
* In any case I can't change the router settings because my isp is weird about it. (I can't even access the router with the usual address).

3.- I re-downloaded brunch stable and chros volteer recovery, just to be safe and I used ext4 again. It worked this time. But the wifi didn't. Same as always

I'm gonna be busy tomorrow so I might not have the chance but I'm gonna have to kill linux and try the single boot option.
Also, here's a long picture of the install, just in case https://imgur.com/a/A846WVB
Also also, quick question: can you install drivers using linux in chros or in the shell by any chance?

tejasraman, 2022-02-14:
NO I used the config menu

tejasraman, 2022-02-14:
Plus, trust sebanc, he is the dev of Brunch.

tejasraman, 2022-02-14:
And also, I am using a tri-boot(PopOS, Arch Linux, and my new ChromeOS). Wi-Fi works fine

tejasraman, 2022-02-14:
It worked with the Linux grub editor too though :)

tejasraman, 2022-02-14:
[image]
Updated to latest unstable, still works great

spvmsolo, 2022-02-14:
Once more, I'm so happy for you
And I do appreciate all the help! XD

spvmsolo, 2022-02-15:
I did the single boot. It didn't work. I'm out of ideas.
So that's the end of my story.

tejasraman, 2022-02-16:
Replace the wifi card :)  8821ce seems to work for me, out of ideas.

spvmsolo, 2024-04-25:
2 years and I tried again. Brunch r123 chros voxel v123 aaaand IT WORKS!
It's fixed! Up and running!

closed 2024-04-25 by spvmsolo