sebanc/brunch #1405: Mic Not Working

ksoham622, 2022-01-31:
I installed Chrome Os with my Windows 11 but my Mic is not working. When I plug in Wired Headphones then the Mic is working but that is the headset mic and internal mic is not working. Mic of Bluetooth headset is also not working. I tried using both Internal_mic_fix options but none are working .

My Laptop Specs:

Lenovo Ideapad 330 (81DC)

Intel Core i3 7th Gen

8 Gb Ram

1 Tb HD

Chrome OS version: 97 and Brunch Version: stable r97
[image]

I did some research and I read this thread https://github.com/sebanc/brunch/issues/56

But that did not work. Then I found this post on another subreddit [alsamixer --card=0](https://www.reddit.com/r/chromeosfans/comments/gnq8pb/possible_fix_to_make_mic_work_in_the_chromeos/l)

alsamixer --card=0

By using the above command I was able to start the internal mic but the mic was not picking up the actual voice it was just picking stuttering sounds.

Please Help.

Everything except the mic is working.

Thanks For making this project.

150mks, 2022-02-01:
Eu tenho o mesmo problema, além desse problema, o notebook é perfeito, eu estarei esperando pelo resultado

Mrid22, 2024-09-23:
which a mixer commands did you use? I have the same issue

closed 2026-01-29 by github-actions[bot]