sebanc/brunch #1411: Stuck At Chromeos-Boot-alert : self repair

Aarshan123, 2022-02-02:
So, I have installed rammus_image file because i have intel core i3 8th gen processor but when i try to open chromeos boot menu through grub2win it just does like read only and then says stuck at chromeos-boot-alert :self repair for a long time.

Aarshan123, 2022-02-02:
also i use a custom code in grub2win that is - 
img_part=/dev/sda3
img_path=/ChromeOS/rammus_image.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
	cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,gpt7)/initramfs.img

Aarshan123, 2022-02-02:
And Also If This Helps My rammus_image.img file is in ChromeOS (E:) under a folder named ChromeOS and its a ntfs file system drive.

ksoham622, 2022-02-03:
I think, you followed a video guide which was old. And there are some changes So follow this guide [Chrome OS Dualboot with windows](https://github.com/sebanc/brunch/blob/master/install-with-windows.md#dualboot-installations).

And download files from this Github page or links provided in there.

ksoham622, 2022-02-03:
And if you have a slow HDD creating the image with WSL2 will take much time you can make LIVE USB with a Linux distro

Aarshan123, 2022-02-03:
> I think, you followed a video guide which was old. And there are some changes So follow this guide [Chrome OS Dualboot with windows](https://github.com/sebanc/brunch/blob/master/install-with-windows.md#dualboot-installations).
> 
> And download files from this Github page or links provided in there.

Thanks! Will Try It!!

Aarshan123, 2022-02-03:
Ok Gonna Reboot And Try It!

Aarshan123, 2022-02-03:
and if it works imma close this thing

ksoham622, 2022-02-03:
Ok

Aarshan123, 2022-02-03:
I Have a problem i did the create user section thing and i booted in the grub menu but what do i do now?

ksoham622, 2022-02-03:
Did you see grub2win ?
And was chrome os and chrome settings present in grub2win ?

Aarshan123, 2022-02-03:
i did see grub2win options there were only 3 options Windows, boot information and utilities and shutdown the system how do i boot into chromeos?

ksoham622, 2022-02-03:
boot into windows again
open grub2win
then go to the partition where your chromeos.img is there and open the txt file which is present with it select and copy all the contents from that txt file.
now go to grub2win manage boot 
add new entry
Select Create user section from the 'Type' section. This will open a text file and paste all the contents you copied from that txt file
Click Ok and Apply to save your entries into Grub2win.
Your entry will not be saved unless you click both Ok and Apply.

Aarshan123, 2022-02-03:
i already did it

ksoham622, 2022-02-03:
do it again

Aarshan123, 2022-02-03:
ok

Aarshan123, 2022-02-03:
u mean this code right? 
menuentry "ChromeOS" --class "brunch" {
	search --no-floppy --set=root --file /brunch/chromeos.img
	loopback loop /brunch/chromeos.img
	source (loop,12)/efi/boot/settings.cfg
	if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug loop.max_part=16 img_uuid=f44b99ee-11b1-4194-b8d8-cb57b6991ab7 img_path=/brunch/chromeos.img \
			console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
	else
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug loop.max_part=16 img_uuid=f44b99ee-11b1-4194-b8d8-cb57b6991ab7 img_path=/brunch/chromeos.img
	fi
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

menuentry "ChromeOS (settings)" --class "brunch-settings" {
	search --no-floppy --set=root --file /brunch/chromeos.img
	loopback loop /brunch/chromeos.img
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug loop.max_part=16 img_uuid=f44b99ee-11b1-4194-b8d8-cb57b6991ab7 img_path=/brunch/chromeos.img
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

ksoham622, 2022-02-03:
If you do it alright you can see chrome os and chrome os settings in grub2win

ksoham622, 2022-02-03:
yes that one

Aarshan123, 2022-02-03:
ok but i dont have a chromeos option tho

ksoham622, 2022-02-03:
earch --no-floppy --set=root --file /brunch/chromeos.img
loopback loop /brunch/chromeos.img
source (loop,12)/efi/boot/settings.cfg
if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params
cros_secure cros_debug loop.max_part=16 img_uuid=f44b99ee-11b1-4194-b8d8-cb57b6991ab7 img_path=/brunch/chromeos.img
console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
else
linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params
cros_secure cros_debug loop.max_part=16 img_uuid=f44b99ee-11b1-4194-b8d8-cb57b6991ab7 img_path=/brunch/chromeos.img
fi
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img

search --no-floppy --set=root --file /brunch/chromeos.img
loopback loop /brunch/chromeos.img
source (loop,12)/efi/boot/settings.cfg
linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1
cros_secure cros_debug loop.max_part=16 img_uuid=f44b99ee-11b1-4194-b8d8-cb57b6991ab7 img_path=/brunch/chromeos.img
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img

ksoham622, 2022-02-03:
remove and add entry again in grub2win

Aarshan123, 2022-02-03:
[image]
this is the options that i have and below down is the create user section

ksoham622, 2022-02-03:
it looks fine
press apply and ok

Aarshan123, 2022-02-03:
ok

Aarshan123, 2022-02-03:
should i add the code that u gave me?

ksoham622, 2022-02-03:
no dont

ksoham622, 2022-02-03:
if you add code copy it from that txt file

Aarshan123, 2022-02-03:
but like how do i boot into chromeos when i dont have the chromeos option?

ksoham622, 2022-02-03:
reboot

Aarshan123, 2022-02-03:
ok i will

Aarshan123, 2022-02-03:
just a sec

Aarshan123, 2022-02-03:
same results only 3 options

ksoham622, 2022-02-03:
if you do not see the options then copy the code in the brackets do not copy anything outside the brackets

ksoham622, 2022-02-03:
search --no-floppy --set=root --file /brunch/chromeos.img
loopback loop /brunch/chromeos.img
source (loop,12)/efi/boot/settings.cfg
if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params
cros_secure cros_debug loop.max_part=16 img_uuid=f44b99ee-11b1-4194-b8d8-cb57b6991ab7 img_path=/brunch/chromeos.img
console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
else
linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params
cros_secure cros_debug loop.max_part=16 img_uuid=f44b99ee-11b1-4194-b8d8-cb57b6991ab7 img_path=/brunch/chromeos.img
fi
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img

ksoham622, 2022-02-03:
so did it work ?

ksoham622, 2022-02-03:
ok so try this idk if this will work but it should probably
img_part=/dev/sda3
img_path=/brunch/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off
cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,gpt7)/initramfs.img

make sure you replace sda3 by the partition you selected
and copy this code into grub2win and in grub2win select add new entry custom code or submenu

Aarshan123, 2022-02-03:
okk

Aarshan123, 2022-02-03:
wait lemme try

ksoham622, 2022-02-03:
did it work ?

Aarshan123, 2022-02-03:
yes

ksoham622, 2022-02-03:
can you see 2 options ?

Aarshan123, 2022-02-03:
the create user thing u told me earlier worked

Aarshan123, 2022-02-03:
yes settings and chrome

ksoham622, 2022-02-03:
> ok so try this idk if this will work but it should probably img_part=/dev/sda3 img_path=/brunch/chromeos.img search --no-floppy --set=root --file $img_path loopback loop $img_path linux [...]

this ?

ksoham622, 2022-02-03:
okay then

Aarshan123, 2022-02-03:
noo

Aarshan123, 2022-02-03:
but wait

Aarshan123, 2022-02-03:
is it normal to be stuck on the patches are being applied thing for like 20 mins

ksoham622, 2022-02-03:
first boot might take time
depending on the system though

Aarshan123, 2022-02-03:
ohh thanks btw do u have discord

Aarshan123, 2022-02-03:
then imma close this thing

ksoham622, 2022-02-03:
yes i have

Aarshan123, 2022-02-03:
whats the user

Aarshan123, 2022-02-03:
mine is Tanjiro Kamado || Sun Hashira ||#2284

ksoham622, 2022-02-03:
join brunch discord

ksoham622, 2022-02-03:
https://discord.gg/x2EgK2M

Aarshan123, 2022-02-03:
ok anyways imma reboot

Aarshan123, 2022-02-03:
see brunch general

Aarshan123, 2022-02-03:
TYSM IM ON CHROME OS RN!!!

closed 2022-02-03 by Aarshan123