sebanc/brunch #1417: Android Apps won't open after relogging, searching for permanent fix

Icey67, 2022-02-04:
[](https://user-images.githubusercontent.com/60461186/152188067-80a3064b-e6f2-45a6-8c51-10de6080d61b.png)image
Everytime when i start ChromeOS, Android container errors keep appearing and will disappear by relogging but i'm sick of it,
Is there anyway i can permanently fix this?

Icey67, 2022-02-04:
The errors always keep changing, this is a screenshot i got from like 2 days ago of running `cat /var/log/messages | grep Android`
[image]

And this is the new output:
`2022-02-03T21:33:31.499634Z ERR session_manager[1305]: ERROR session_manager: [android_oci_wrapper.cc(159)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2022-02-03T21:33:31.508700Z ERR session_manager[1305]: ERROR session_manager: [session_manager_impl.cc(1962)] Starting Android container failed.
2022-02-03T21:33:31.508721Z ERR session_manager[1305]: ERROR session_manager: [dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2022-02-03T21:33:34.299129Z INFO session_manager[1305]: INFO session_manager: [session_manager_impl.cc(1968)] Started Android container with PID 3747
2022-02-03T21:33:34.316721Z INFO session_manager[1305]: INFO session_manager: [session_manager_impl.cc(1455)] Android container is running with PID 3747
2022-02-03T22:41:41.931489Z INFO session_manager[1305]: INFO session_manager: [session_manager_impl.cc(2059)] Android container with PID 3747 stopped
2022-02-03T22:41:41.932285Z ERR session_manager[1305]: ERROR session_manager: [session_manager_impl.cc(1533)] Error getting Android container pid.
2022-02-03T22:41:41.932295Z ERR session_manager[1305]: ERROR session_manager: [dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerShutdownFail, Message=Error getting Android container pid.
2022-02-04T08:11:31.652889Z ERR session_manager[1299]: ERROR session_manager: [android_oci_wrapper.cc(159)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2022-02-04T08:11:31.662596Z ERR session_manager[1299]: ERROR session_manager: [session_manager_impl.cc(1962)] Starting Android container failed.
2022-02-04T08:11:31.662648Z ERR session_manager[1299]: ERROR session_manager: [dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2022-02-04T15:15:10.527354Z INFO session_manager[22257]: INFO session_manager: [session_manager_impl.cc(1968)] Started Android container with PID 22524
2022-02-04T15:15:15.744315Z INFO session_manager[22257]: INFO session_manager: [session_manager_impl.cc(1455)] Android container is running with PID 22524
2022-02-04T18:05:06.405335Z INFO session_manager[22257]: INFO session_manager: [session_manager_impl.cc(2059)] Android container with PID 22524 stopped
2022-02-04T18:05:57.392197Z ERR session_manager[22257]: ERROR session_manager: [android_oci_wrapper.cc(159)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2022-02-04T18:05:57.394349Z ERR session_manager[22257]: ERROR session_manager: [session_manager_impl.cc(1962)] Starting Android container failed.
2022-02-04T18:05:57.394361Z ERR session_manager[22257]: ERROR session_manager: [dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2022-02-04T18:06:34.566654Z ERR session_manager[27855]: ERROR session_manager: [android_oci_wrapper.cc(159)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2022-02-04T18:06:34.574243Z ERR session_manager[27855]: ERROR session_manager: [session_manager_impl.cc(1962)] Starting Android container failed.
2022-02-04T18:06:34.574296Z ERR session_manager[27855]: ERROR session_manager: [dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2022-02-04T18:07:25.409072Z ERR session_manager[27855]: ERROR session_manager: [android_oci_wrapper.cc(159)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2022-02-04T18:07:25.417906Z ERR session_manager[27855]: ERROR session_manager: [session_manager_impl.cc(1962)] Starting Android container failed.
2022-02-04T18:07:25.417959Z ERR session_manager[27855]: ERROR session_manager: [dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2022-02-04T18:10:00.812801Z ERR session_manager[29980]: ERROR session_manager: [android_oci_wrapper.cc(159)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2022-02-04T18:10:00.820665Z ERR session_manager[29980]: ERROR session_manager: [session_manager_impl.cc(1962)] Starting Android container failed.
2022-02-04T18:10:00.820716Z ERR session_manager[29980]: ERROR session_manager: [dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2022-02-04T18:10:51.717844Z ERR session_manager[29980]: ERROR session_manager: [android_oci_wrapper.cc(159)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2022-02-04T18:10:51.722386Z ERR session_manager[29980]: ERROR session_manager: [session_manager_impl.cc(1962)] Starting Android container failed.
2022-02-04T18:10:51.722407Z ERR session_manager[29980]: ERROR session_manager: [dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.`

ksoham622, 2022-02-05:
Did Android apps worked on first boot ?

Icey67, 2022-02-06:
> Did Android apps worked on first boot ? Try android_init_fix from framework. And when you see the login screen wait for 5 minutes and then login.your apps will work.

The android apps did work on the first boot, i already checked android_init_fix but waiting on the login screen? Why? Also this won't make it a permanent fix because then I would always have to wait 5 mins on the login screen Everytime I boot chrome os

ksoham622, 2022-02-06:
That is the only fix available try it and tell me

Icey67, 2022-02-06:
alright i'm gonna try it

Icey67, 2022-02-06:
i just waited 5 mins and the laptop just shut down

3bagorion33, 2024-05-09:
I am experiencing the same thing with the Coral image on Intel z3795, build 123 with brunch 123. I tried using Rammus, but it doesn't load the graphics at all. I read in another thread that it's due to a missing i915 graphics driver. The first clean start opens the Google Play app with error 8. I have also tried disabling and re-enabling the Play Market feature. After that the user agreement window appears, but after accepting it - error 8 again.

I tried to enable the fix in the kernel, reboot and wait 5 minutes - error 8 again.

3bagorion33, 2024-05-09:
So, the latest news from the battlefield. I was able to run ChromeOS from the image from the description below this video. https://www.youtube.com/watch?v=puZ_8rPEVbI I used the image for Gen 3 and higher processors. It uses build 120.0.6099.272, which according to the table at https://chromiumdash.appspot.com/serving-builds?deviceCategory=ChromeOS means quandiso. I will now try to enable the update and/or make my own build based on the latest quandiso build.

3bagorion33, 2024-05-09:
Next research: I found information that this error occurs when there is no virtualization. 

According to this information https://stackoverflow.com/questions/11116704/check-if-vt-x-is-activated-without-having-to-reboot-in-linux output gives 5, which means virtualization is enabled. I ran Linux Mint on my tablet and got output 5. The output of the kvm-ok command also confirms that virtualization is enabled:
```
INFO: /dev/kvm exist
KVM acceleration can be used
```

ZProLegend007, 2024-08-13:
Also having that issue on Coral image, @3bagorion33. I will try the Nissa image which is the image for the quandiso board (more generic image). I'm doing this on a 2014 MacBook Air.

closed 2026-01-28 by github-actions[bot]