sebanc/brunch #1419: trick : how to enable only bluetooth on a wifi+bluetooth realtek rtl8723bu card

pierro78, 2022-02-06:
Got a netbook ( see https://www.reddit.com/r/Brunchbook/comments/s5p0lf/chromeos_via_brunch_is_great_on_cheap_netbooks/ or https://forum.hardware.fr/hfr/OrdinateursPortables/Netbook/3000x2000-plastique-302x220x16mm-sujet_87894_1.htm ) with an internal wifi+bluetooth realtek rtl8723bu card. 
This  wifi+bluetooth realtek rtl8723bu is quite infamous (see https://github.com/morrownr/USB-WiFi) for having a bad linux/brunch-chromeos support (I noticed very weak wifi reception in linux and in brunch-chromeos it would even not connect my router's wifi)

**Here is a trick if you want to use only the BT of the rtl8723bu** : just install (in brunch-chromeos settings) the driver for the RTL88x2BU chipset : then the BT of the rtl8723bu will work but not the wifi ! 👍 

(this solves my issue on my netbook to use the wifi of a tiny rtl8188fu USB dongle (much better than the rtl8723bu) and then getting BT to work with my BT speaker ...)

(there might be simple linux commands to also do that but I (accidently) found out that this method is quite easy to apply ;) ...)

ksoham622, 2022-02-06:
Hello
Don't post this on issue put it on discussion.

pierro78, 2022-02-06:
> Hello Don't post this on issue put it on discussion.
Oops didn t notice the "Discussions" tab, sorry ... here is the discussion : https://github.com/sebanc/brunch/discussions/1421